ros-industrial/industrial_ci #13: bloom-release industrial_ci?

130s, 2016-01-05:
Idea triggered at https://github.com/ros-industrial/industrial_ci/pull/12#issuecomment-169051181 but I've been wanting to ask for the opinion.

There's no Technical issue in bloom-releasing the repo/package `industrial_ci`, since it is just a set of bash and python scripts (so far). I can work on this if there's an enough advantage or requests, which I don't see by myself so far. All I can think of now is:

Advantage
- Free the client repo maintainers from git command; 
 - NOW recommended way is `git clone %URL%`
 - NEW way may become `apt-get install ros-indigo-industrial-ci`
- We're able to nicely force the version to be used

Disadvantage
- More maintenance work

gavanderhoorn, 2016-01-05:
Other potential disadvantage: the ROS deb repositories support only a single version of packages. It's a hypothetical use case, but with the current approach it's relatively easy to check out an older revision/tag.

Obviously those wanting to do that could use the git repository anyway, but then, why release the pkgs? As @130s writes: it would be more maintenance work.

The majority of developers will however probably only use the 'normal way' (ie: use latest version).

I'm +0 on this one.

shaun-edwards, 2016-01-05:
If you release this package, how will we call the shell scripts?  Do you use `rosrun` or simply call the script using the install path?  If there is any change to how the scripts are called, won't this be an incompatible change with those that clone the repo?

130s, 2016-01-06:
> but then, why release the pkgs?

Yeah, that was the big question, and it remains so after hearing from you two ;)

> If you release this package, how will we call the shell scripts?

I see, that's another concern. `rosrun` could workaround this issue nicely (as long as all the scripts are properly installed), but again, I don't see any motivation to make this change and cause changes in downstream repositories.

Thank you for the opinions, now I'm convinced to close this without any action.

closed 2016-01-06 by 130s