ros-industrial/industrial_ci #17: License of the package

130s, 2016-01-12:
I thought I would change the license in a trial to conform ROS-I recommendation (apache 2.0), but the repository I got much inspiration from (`jsk_travis`) is BSD (and some scripts from there are simply redistributed here), I like to hold back.

If there's any other opinions, please reopen and discuss.

130s, 2016-01-16:
Just happened to find [Apache2.0 LICENSE file](https://github.com/ros-industrial/industrial_ci/blob/master/LICENSE), which Shaun created when he initiated this repo. I/we gotta figure out what license we use.

gavanderhoorn, 2016-05-27:
IANAL, but AFAIK, you can relicense artefacts, but you'll have to get permission from the original authors.

Also: it's not a requirement to have all artefacts under the same license. It's perfectly possible for some bits to be BSD, with others under Apache 2. And apparently BSD and Apache 2 are very compatible with each other.

Perhaps @ipa-mirb can refer us to some relevant documents?

shaun-edwards, 2016-05-31:
I have the same understanding as @gavanderhoorn.  I would propose that we keep existing licenses as is and require that any new "code" be licensed under Apache 2.0

ipa-mirb, 2016-06-07:
Unfortunately I do not have specific documents to point you to, as last time we dealt with licensing at the "OSS - legal aspects" event here at IPA we discussed more general terms. But from what I understood mixing BSD and Apache should not be a problem (whereas e.g., LGPL and BSD could be, I am afraid).

130s, 2017-01-20:
As I noted in https://github.com/ros-industrial/industrial_ci/pull/108#issuecomment-274114279, now there are only files that were created within this repo so that we have full control of the license for this package/repo (INAL but I hope so). So in https://github.com/ros-industrial/industrial_ci/pull/113 I changed the package license to Apache 2.0.

closed 2017-01-20 by 130s