ros-industrial/industrial_ci #21: Adapt catkin_make?

130s, 2016-02-10:
From https://github.com/ros-industrial/industrial_core/pull/136#issuecomment-182246945

> As @ipa-mdl already commented in ros-industrial/ros_canopen#155, this can / should be reverted once catkin_tools 0.4.0 is released. I'd rather we switch to catkin_make for the time being (don't know how feasible that is though, @130s?).

It's absolutely feasible and in fact I thought I offered to do so somewhere (I just can't find it now, but it's not a big deal), since I intend `industrial_ci` to work with multiple build tools (`catkin_make`, `ament` etc.) I don't think I'll be able to work on this this week though.

gavanderhoorn, 2016-02-10:
That would probably then be configurable through the `BUILDER` env var?

ipa-mdl, 2016-02-10:
I am not sure if `catkin_make` is the right way to go, since it will not find bugs in dependencies etc. that might prevent  release. `catkin_make_isolated --merge` might be a better option.

Another possible solution is to use latest catkin_tools from source in the repositories that need it
This could be interluded directly in .travis.yml: https://github.com/catkin/catkin_tools/blob/master/.travis.yml#L35

gavanderhoorn, 2016-02-10:
@ipa-mdl wrote:
> I am not sure if `catkin_make` is the right way to go, since it will not find bugs in dependencies etc. that might prevent release. `catkin_make_isolated --merge` might be a better option.

I agree about misconfigured dependencies going unnoticed, but the context for my comment was that we're now changing *our* files to work-around a bug in a build tool not under our control. If I understand @130s correctly, he wanted to add support for additional build tools anyway, so if that could be exploited to add a temporary work-around for this particular issue with `catkin_tools`, I'd find that more palatable than what ros-industrial/industrial_core#136 is proposing.

Obviously, if there is support for additional builders, we could always decide to use `catkin_make_isolated` as well.

> Another possible solution is to use latest catkin_tools from source in the repositories that need it
This could be interluded directly in .travis.yml: https://github.com/catkin/catkin_tools/blob/master/.travis.yml#L35

Hm. That would mean that we're going to roll out a CI setup that is tracking a development repository for its main build tool. I'm not sure I'd be so happy about that.

ipa-mdl, 2016-02-10:
> Obviously, if there is support for additional builders, we could always decide to use `catkin_make_isolated` as well.

I am a little bit concerned that it bloats the code. (Too many `$BUILDER` branches..)

> Hm. That would mean that we're going to roll out a CI setup that is tracking a development repository for its main build tool. I'm not sure I'd be so happy about that.

I don't like this either. However, it would just affect the testing process in the respective repositories and nothing else, especially it would not mess with the package dependencies.
And it is a quick solution, available right now.

~~Another quick fix might be to source devel/setup.bash before running the tests, but I haven't tested it.~~ (It is already include and does not work.)

130s, 2018-03-13:
My team (finally) hit the case where `catkin_make` fails while `catkin_tools` passes the build (https://github.com/ros/catkin/issues/925). Because my team takes catkin_make as an official tool at this moment, `industrial_ci` not picking up the issue like that might become a bit of worry. But I don't think I'm willing to re-vitalize the support for catkin_make like https://github.com/ros-industrial/industrial_ci/pull/23.

Any good idea to work this around?

gavanderhoorn, 2018-03-13:
Looking at the comment by Dirk (https://github.com/ros/catkin/issues/925#issuecomment-372712209) I don't see something to work around?

If you need your builds to work under `catkin_make`, you'll have to add `add_dependencies(..)`, as there is not magic way to figure that out automatically, and they are needed to correctly sort the build-dep tree.

`catkin_tools` (and `catkin_make_isolated`) work differently: topological sorting is done based on package boundaries, so packages are fully built before others are started. A side-effect of this is that `add_dependencies(..)` is not needed (in all cases) as CMake / Catkin has enough info to figure out a proper build order.

If your CTO doesn't want to vet the use of `catkin_tools`, but you don't want to have to add `add_dependencies(..)` manually, perhaps using `catkin_make_isolated` would be an option.

ipa-mdl, 2019-05-15:
Support for `catkin_make` and `catkin_make_isolated` is implemented in #361

closed 2019-08-18 by ipa-mdl