ros-industrial/industrial_ci #24: "What to be checked" section may not be clear enough

130s, 2016-03-03:
I've moved from https://github.com/ros-drivers/freenect_stack/pull/27#issuecomment-191342685
> I am not sure exactly what is involved or how it all works, am very supportive of the concept.

@jack-oquin Thanks for the comment!

I've tried my best to [describe what checks are carried out](https://github.com/ros-industrial/industrial_ci#what-are-checked) with the current config, but I'm not yet sure if that portion of document actually makes sense / easy to understand due to my English / choice of terminologies / engineering soul etc. Since that is the only part that explains the "features" that this package provides, I'm willing to add any possible improvements. 

I appreciate if anyone gives any feedback.

jack-oquin, 2016-03-11:
> If tests in designated downstream packages pass.

The word ``downstream`` seems to have become ambiguous these days when discussing building and packaging. Traditionally, the main source repository was considered ``upstream``, while forks, derived Debian source packages and finally Debian binaries were farther and farther ``downstream``. 

Lately, I've heard lots of people using those terms to describe dependencies and reverse dependencies with other packages, which I find confusing.

So, are these tests run using packaged Debian binaries? Or, are they testing whether ROS source packages that depend on this one build cleanly, and that their unit tests still work?

130s, 2016-04-19:
Jack, thank you for the comment. 
In #34, I added a terminology section for clarification, but to answer your question,

> are these tests run using packaged Debian binaries? 

For the depended (aka *upstream*) packages, yes by default.

> Or, are they testing whether ROS source packages that depend on this one build cleanly, and that their unit tests still work?

If you specify dependent (*downstream*) packages, yes.

Hope these difference is a bit clearer with #34...

130s, 2016-06-21:
Hope #34 addresses the concern here. Please reopen or comment if not.

closed 2016-06-21 by 130s