ros-industrial/industrial_ci #26: Nodelet dependency

ipa-mdl, 2016-04-12:
The `ros_canopen` travis jobs currently work if tested against the indigo repo, but fail against the shadow-fixed repo (https://travis-ci.org/ros-industrial/ros_canopen/builds/122520707).

It's all good until the test install space step, which fails with
````
+++rospack plugins --attrib=plugin nodelet
[rospack] Error: no such package nodelet
```
The corresponding line is https://github.com/ros-industrial/industrial_ci/blob/master/travis.sh#L233
It introduces a dependency on the `nodelet` package.

For some reason it worked without having the `nodelet` package before (transitive dependency?). However, this seems to have changed with new changes in shadow-fixed.

Where does this line come from? What is its purpose?
I would suggest to remove it.

ipa-mdl, 2016-04-12:
I believe if fails (which is correct!) because of https://github.com/ros/rospack/pull/51 which was released one month ago, but not yet synced.

130s, 2016-04-12:
Sorry for the inconvenience. In fact I wasn't aware of that error since all repos I maintain that uses `industrial_ci` seem not to have this issue (, which means that there is at least one ROS package that contains `nodelet` plugin). I'm testing a patch now for this.

130s, 2016-04-13:
> Why do we need this line at all?

 In order to catch implicit mistake that can occur when writing packages using `nodelt` plugins.

 As I kind of explained in https://github.com/ros-industrial/industrial_ci/pull/27#issue-147890645, often when a package uses `nodelet` plugins, they fail to be recognized by catkin due to misconfiguration, which developers won't notice until runtime where it tends to be not the easiest to track down the root cause at that time. 
This rospack line is intended to catch the error at the compilation time.

 **Execuse**
 But yes, due to the bug in `rospack` that was fixed in https://github.com/ros/rospack/pull/51, this line hasn't really done the job for any users.

 Now, with https://github.com/ros/rospack/pull/51, the line returns an error as originally intended. Problem is that, as we're seeing, it's too aggressive so that testing packages without dependency to nodelet fails.

 **Workaround**

 With the temporary workaround suggested in https://github.com/ros-industrial/industrial_ci/pull/27:
 - rospack line is still run.
 - Travis no longer returns with failure, even when rospack line fails.
   - Ideally Travis returns with failure when rospack line fails, but for implementing this I need more time.

> Which ROS-I packages really need this test?

Unfortunately I'm not keeping track of which packages need what tests.
(I think packages commonly used like this can have more features than each client package wants to use.)

ipa-mdl, 2016-04-13:
`industrial_ci` is about CI and thus automatic tests.
`rospack plugins --attrib=plugin nodelet` is not a test, it just for debugging.
(With the new `rospack` behaviour it tests if `nodelet` is installed..)

I don't like the workaround #27 because it displays a big warning whout any need for it,

I see 3 alternatives:
1. remove the line here, and add it to the respective packages
2. execute the line only in case of test failures
3. install the nodelet dependency beforehand

> Unfortunately I'm not keeping track of which packages need what tests.

A [quick search](https://github.com/search?utf8=%E2%9C%93&q=org%3Aros-industrial+nodelet+language%3AXML+extension%3Axml+language%3AXML&type=Code&ref=advsearch&l=XML&l=XML]) revealed that only `human_tracker`, `industrial_ci` and `industrial_pcl_experimental` install nodelets.

130s, 2016-04-13:
Ok, I don't have much time unfortunately now for further discussion, but this issue is critical (prevents users from using the repo) so I made a quick path in #28. It just removes `rospack pluging` line. @ipa-mdl please leave approval there as ros-i repository requires +1s for merge.

closed 2016-04-13 by 130s

ipa-mdl, 2016-04-13:
Thanks for the quick fix!

I just came up with another alternative:
We could add another variable with whitespace-separated plugin bases (defaults to empty string?).
Then the plugin settings could be listed for each of the entry.

If you like it, I can provide an implementation.
I am just not sure if it's worth the effort..

130s, 2016-05-12:
Ok, it'd be nice to get back the plugin check in.

> I just came up with another alternative:
> We could add another variable with whitespace-separated plugin bases (defaults to empty string?).
> Then the plugin settings could be listed for each of the entries.
> 
> If you like it, I can provide an implementation.
> I am just not sure if it's worth the effort..

Could you show me a sample (can be pseudo)?

Maybe we might as well open a new ticket for it.

ipa-mdl, 2016-05-12:
> Could you show me a sample (can be pseudo)?

Something like
```bash
for p in $TEST_PLUGINS_FOR; do
  rospack plugins --attrib=plugin $p
done
```

Example: `TEST_PLUGINS_FOR="nodelet controller_interface"`
if unspecified nothing happens.

commit 9d08a556 (2016-04-13): Temporarilly disable rospack plugin line (fixes #26).
commit efd92365 (2016-04-13): Merge pull request #28 from 130s/fix/rospackplugins_workaround

Temporarilly disable rospack plugin line (fixes #26).