ros-industrial/industrial_ci #29: Reducing build time for VTK 7 and PCL trunk

130s, 2016-04-15:
From [the maling list](https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/rpGkmKlBebs) by @VictorLamoine

>> I would like to use indusrial_ci for the bezier project.
>> However the dependencies are VTK 7 and PCL trunk so I need to compile/install those on the Travis server.
>> This takes time and exhausts the available time for "free" builds.
>> 
>> See for example:
>> https://travis-ci.org/InstitutMaupertuis/bezier/jobs/123141945
>> 
>> I reduced the PCL build to a minimal build but it still takes a lot of time.
>> There are some PPA for VTK nightly but no PPA for PCL trunk.
>> 
>> Is there a way to speed up this? Providing binaries?

@shaun-edwards 

> Do you need to build PCL/VTK every time or would weekly/monthly binaries be OK?  If so, then I could imagine that you modify the Travis script to download the binaries/debians from some repo (PPA, github, ftp, etc...) and install them.  You would still be on the hook to recompile PCL/VTK every so often, but I'm guessing you aren't worried about those libraries causing issues.

130s, 2016-04-15:
Honestly I may not have many solution options to this problem.

Provided that using binary is acceptable as Shaun asked, you can specify additional binaries by `ADDITIONAL_DEBS` (see also [readme](https://github.com/ros-industrial/industrial_ci/blob/master/README.rst#optional-environment-variables)) in the env matrix in your `.travis.yml`. E.g.,

```
env:
  matrix:
    - USE_DEB=true
      ROS_DISTRO="indigo"
      ROS_REPOSITORY_PATH=http://packages.ros.org/ros/ubuntu
      ADDITIONAL_DEBS ros-indigo-pcl-ros
```

(I'm not sure if the above code installs what you need though).

VictorLamoine, 2016-04-15:
I managed to prepare a `.deb` of VTK 7.0.0 pretty easily;3
https://github.com/VictorLamoine/vtk7_travis_test
https://travis-ci.org/VictorLamoine/vtk7_travis_test/builds/123279567

I'll try to do the same for PCL and get back to you when I have something working (or I give up :smile:)

VictorLamoine, 2016-04-18:
From what I have tested the VTK `.deb` is working fine.

I could not build a `.deb` for PCL because it doesn't work (http://www.pcl-developers.org/PCL-trunk-PPA-td5710079.html) so I created a `.zip` archive instead. This is 'dangerous' because it doesn't manage the dependencies at all.

It seems to be working fine:
https://travis-ci.org/VictorLamoine/pcl1.8_travis_test

However when using the same binaries I end up with [missing symbols when linking](https://travis-ci.org/InstitutMaupertuis/ensenso_extrinsic_calibration/jobs/123889359#L6985) (these symbols are [not missing](https://travis-ci.org/InstitutMaupertuis/ensenso_extrinsic_calibration/jobs/123897370#L657)).

VictorLamoine, 2016-04-18:
Update: now [ensenso_extrinsic_calibration](https://travis-ci.org/InstitutMaupertuis/ensenso_extrinsic_calibration/branches) is working fine with Travis.
I still have problems with missing sympols on [Bezier](https://travis-ci.org/InstitutMaupertuis/bezier/builds/123930353), I think because of VTK CMake configuration, not sure yet.

I have created https://github.com/InstitutMaupertuis/advanced_industrial_ci to help other users facing the same issue.

130s, 2016-04-19:
@VictorLamoine Glad that you managed to work around your problem.

I looked at [InstitutMaupertuis/advanced_industrial_ci](https://github.com/InstitutMaupertuis/advanced_industrial_ci) repo and tried to come up with more generic way of testing to be implemented in `industrial_ci`. But what `InstitutMaupertuis/advanced_industrial_ci` does seems to me a reasonable enough approach.

PR #62 (2016-06-06): Provide instructions for advanced_industrial_ci
Fixes #29 
I think the scripts are good enough, everything works across several repositories.
closed 2016-06-06 by 130s