ros-industrial/industrial_ci #30: catkin clean -a

Jmeyer1292, 2016-04-19:
After merging a new Descartes pull request, I received a notification that the build had failed. A quick search reveals:

```
+++catkin clean -a

usage: catkin [-h] [-a] [--test-colors] [--version]

              [--force-color | --no-color]

              [build | clean | config | create | env | init | list | locate | profile]

              ...

catkin: error: unrecognized arguments: -a
```
See the following for the full log:
https://travis-ci.org/ros-industrial-consortium/descartes/jobs/124093593

130s, 2016-04-19:
Thanks for the report. I'll look into this.
Seems related https://github.com/catkin/catkin_tools/issues/320.

130s, 2016-04-19:
Should be resolved by #31. @Jmeyer1292 @VictorLamoine if it isn't please let me know on this ticket.

closed 2016-04-19 by 130s

130s, 2016-04-19:
You can simply re-run the Travis jobs that failed.