ros-industrial/industrial_ci #36: catkin_tools_prebuild:cmake step fails if the Travis CI language is "cpp"

pjreed, 2016-04-25:
I have a few repositories that I want to build with both clang and gcc.  In order for the `compiler:` value in the .travis.yml file to work, the `language:` needs to be set to `cpp` rather than `generic`; otherwise it will only compile using gcc.  However, If I make that change, the build fails almost immediately.  When cmake is running for the first package, it prints out this error:

```
[catkin_tools_prebuild:cmake] -- Found gtest sources under '/usr/src/gtest': gtests will be built
[catkin_tools_prebuild:cmake] -- Using Python nosetests: /usr/bin/nosetests-2.7
[catkin_tools_prebuild:cmake] ImportError: "from catkin_pkg.package import parse_package" failed: No module named catkin_pkg.package
[catkin_tools_prebuild:cmake] Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
[catkin_tools_prebuild:cmake] CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
[catkin_tools_prebuild:cmake]   execute_process(/opt/python/2.7.10/bin/python
[catkin_tools_prebuild:cmake]   "/opt/ros/indigo/share/catkin/cmake/parse_package_xml.py"
[catkin_tools_prebuild:cmake]   "/opt/ros/indigo/share/catkin/cmake/../package.xml"
[catkin_tools_prebuild:cmake]   "/home/travis/ros/ws_marti_common/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
[catkin_tools_prebuild:cmake]   returned error code 1
[catkin_tools_prebuild:cmake] Call Stack (most recent call first):
[catkin_tools_prebuild:cmake]   /opt/ros/indigo/share/catkin/cmake/catkin_package_xml.cmake:63 (safe_execute_process)
[catkin_tools_prebuild:cmake]   /opt/ros/indigo/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
[catkin_tools_prebuild:cmake]   /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
[catkin_tools_prebuild:cmake]   CMakeLists.txt:4 (find_package)
```

If I set the language back to generic, the error goes away.  In the build log, it does look like the python-catkin-pkg package was installed, so I'm not sure why cmake can't find it.  Here's a complete example of a build log: https://travis-ci.org/pjreed/marti_common/jobs/125634383

130s, 2016-04-25:
Thanks for the report!
Glad that you managed to work around the issue (I believe in https://github.com/swri-robotics/marti_common/pull/318). Although my workload this week is limited, I feel like implementing something permanent to `industrial_ci` for this issue (code, or doc).

Travis CI does special thing with `language: cpp` for the convenience for `Autotools` and `make`, but I'm not sure how that effects cmake-based `catkin`.

https://docs.travis-ci.com/user/languages/cpp
> Default Test Script #
> 
> Because C++ projects on travis-ci.org assume Autotools and Make by default, naturally, the default command Travis CI will use to run your project test suite is
>
>    ./configure && make && make test
>
> Projects that find this sufficient can use a very minimalistic .travis.yml file:
>     language: cpp

ipa-mdl, 2016-04-26:
If I remember it correctly the "language: cpp" pulls in a wrong/incompatible python version.

> otherwise it will only compile using gcc

Can't you just install another compiler and set the CC/CXX environment variables?
This way I am using clang on my machines.

pjreed, 2016-04-26:
I can, although being able to use the `compiler:` setting is nice because Travis CI will generate a build matrix from all of the permutations of that and all of the other environment variables I've set in the `matrix:` section.  I can work around it, it just means having to manually add gcc and clang variations of all of the existing matrix lines.

ipa-mdl, 2017-12-18:
This should be resolved because all command are run in docker now.
However, the build matrix feature cannot be used.

closed 2017-12-18 by ipa-mdl