ros-industrial/industrial_ci #44: pkg-config Installed By Default in Travis Environment, not on OSRF Build Farm

evenator, 2016-05-13:
We recently had a package that required pkg-config as a build tool, but we failed to add it as a dependency in the package.xml. It built fine on developer machines (because we all have pkg-config) and on Travis (because it has pkg-config), but then failed on jenkins.ros.org (which does not have pkg-config installed).

Would it be possible to remove pkg-config from the Travis environment to more accurately reflect the OSRF farm environment?

ipa-mdl, 2016-05-13:
This dependency is pulled in by `rospack`, which is used several times for printing debug information throughout the script.
Have you tried the [prerelase test option] (https://github.com/ros-industrial/industrial_ci#optional-run-ros-prerelease-test)?

130s, 2016-05-16:
Out of curiosity (and possible future enhancement on industrial_ci), could you link to the build error on jenkins you saw?

pjreed, 2016-05-16:
The issue here was a little tricky for us to figure out.  Here's the build that failed; it was the swri_transform_util package in the marti_common repository:
http://build.ros.org/job/Jbin_uT64__swri_transform_util__ubuntu_trusty_amd64__binary/7/

The reason it failed is because the linker couldn't find various yaml-cpp references that were transitively required by swri_yaml_util.

Next, take a look at the build for swri_yaml_util:
http://build.ros.org/job/Jbin_uT64__swri_yaml_util__ubuntu_trusty_amd64__binary/5/

The build succeeded, but because pkg-config was not included in its dependencies, it wasn't installed, and so the appropriate linker flags did not get passed on to other packages that depended on it.

ipa-mdl, 2017-03-20:
This was fixed in #108

closed 2017-03-20 by ipa-mdl