ros-industrial/industrial_ci #49: [Prerelease Test] Test often disregards the actual result from Catkin

130s, 2016-05-16:
For example https://travis-ci.org/anqixu/ueye_cam/jobs/130600243#L9129
```
+catkin_test_results --verbose
Summary: 0 tests, 0 errors, 0 failures
+echo 'ROS Prerelease Test went successful.'
ROS Prerelease Test went successful.
+exit 0
+travis_time_end
+set +x
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function show_testresult took 0 min 0 sec
+cd /home/travis/build/anqixu/ueye_cam
+pwd
/home/travis/build/anqixu/ueye_cam
+++catkin_test_results build
Test results directory "/home/travis/ros/ws_ueye_cam/src/build" does not exist
+++error
+++travis_time_end 31
+++set +x
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function prerelease_from_travis_sh took 18 min 25 sec
```

Looks similar to https://github.com/ros-industrial/industrial_ci/pull/35#issuecomment-216399634 but I'm not sure.

I'm afraid that this issue makes Prerelease Test feature of `industrial_ci` hard to use, since it makes its result unstable regardless the actual test result from Catkin.

ipa-mdl, 2016-05-17:
Why is the output checked in  https://github.com/ros-industrial/industrial_ci/blob/master/ros_pre-release.sh#L74 AND https://github.com/ros-industrial/industrial_ci/blob/master/travis.sh#L175?
The latter incovation is wrong because it includes the non-existing build directory.

130s, 2016-05-18:
Ok, for some reason the Travis job for [the PR](https://github.com/ros-industrial/industrial_ci/pull/35) where Prerelease feature was initially added [didn't run the right code and the job passed without error](https://s3.amazonaws.com/archive.travis-ci.org/jobs/128151566/log.txt) -- it didn't run "[catkin_test_results build](https://github.com/ros-industrial/industrial_ci/pull/35/files#diff-f9a62833538ab14fff849c2be16ab986R177)" that WAS included in the PR.

Anyways, I'm working of a fix.

commit d56e13b8 (2016-05-31): [Prerelease] Remove spurious line to fix #49.
closed 2016-05-31 by 130s