ros-industrial/industrial_ci #53: Hash sum mismatch

VictorLamoine, 2016-05-27:
This is external to `industrial_ci` but all current builds fails because of a server problem:

```
W: Failed to fetch http://dl.hhvm.com/ubuntu/dists/trusty/main/binary-amd64/Packages  Hash Sum mismatch
E: Some index files failed to download. They have been ignored, or old ones used instead.
++++error
++++travis_time_end 31
```

130s, 2016-05-27:
Could you give us a link to some example results? I can't seem to find one.

VictorLamoine, 2016-05-27:
https://travis-ci.org/InstitutMaupertuis/universal_robot_motion/jobs/133313579#L242

VictorLamoine, 2016-06-01:
Closing as the builds are clear from this temporary bug now.
This issue was just to warn other users.

closed 2016-06-01 by VictorLamoine

gavanderhoorn, 2016-06-02:
Docker is having some issues (see docker/docker#23203), so builds failing due to hash mismatches will happen again I reckon.