ros-industrial/industrial_ci #54: Consider building to install space directly

130s, 2016-05-27:
Suggested at https://github.com/ros-industrial/industrial_ci/pull/48#issuecomment-219294989 by @rhaschke
> Consider building to `install` space directly. I don't see a reason why to build twice, for `devel` and `install` space. This only doubles the response time of the test.

gavanderhoorn, 2016-05-27:
This makes sense, I agree with @rhaschke.

130s, 2016-06-04:
One usecase I can think of is that packages are not planned to be released so no need for `install` space either. In that case keeping to use `devel` space might be more convenient and less confusing because most of the tutorials use `devel`. 

But I guess there may not be many users who really need to stick to devel space. So +1

gavanderhoorn, 2016-06-04:
I seem to remember the 'official policy' is to recommend using `install` space as much as possible, especially with `catkin_tools`. The `devel` space is kept only for a measure of backwards compatibility.

Packages failing because they have no `install(..)` statements is a good thing I think: missing those will also cause problems when / if people try to release those packages through the build farm, so catching that early seems like a good thing to do.

rhaschke, 2016-06-04:
> Packages failing because they have no |install(..)| statements is a 
> good thing I think: missing those will also cause problems when / if 
> people try to release those packages through the build farm, so 
> catching that early seems like a good thing to do.
>
I can only underline this argument.

davetcoleman, 2016-07-07:
+1 to install space by default

130s, 2016-09-12:
(Hopefully) fixed by https://github.com/ros-industrial/industrial_ci/pull/67, where I accidentally (sorry...) included the commits that try to implement the fix to this.

130s, 2017-03-22:
https://github.com/ros-industrial/industrial_ci/issues/54#issuecomment-246508323
> (Hopefully) fixed by #67, where I accidentally (sorry...) included the commits that try to implement the fix to this.

I don't think my statement above is correct; #67 doesn't add anything relevant.

I think https://github.com/ros-industrial/industrial_ci/pull/150 fixes this.

130s, 2017-03-26:
Reopening for a discussion.

As a package with "industrial" prefix, requiring `install` space sounds reasonable. Users are pros, advanced grad students...

Meanwhile, when we want to escalate this package to general ROS level as we discussed (e.g. at https://github.com/ros-industrial/industrial_ci/issues/75#issuecomment-267989567), we'll probably want to keep `devel` space usage. I can hardly imagine novice users and research students willing to use `install` space in their own software when they have no future plan to make an installable release. No?

ipa-mdl, 2017-03-26:
> we'll probably want to keep devel space usage

No. The install option just runs the provided install rules on top of the build instructions.
So if you do not specify any rules, your code is not affected at all.
If you specify rules, they get tested.

> I can hardly imagine novice users and research students willing to use install space in their own software when they have no future plan to make an installable release

They don't have to, but it is recommended anyway.
As soon as projects start growing you have to take care of dependencies and package relations.
If someone does not care about these topics, CI testing is not the way to go.

130s, 2017-03-26:
Hm, I can't find where the `install space` is recommended, so [asked](https://groups.google.com/forum/#!topic/ros-sig-buildsystem/OCtNc83s6GA).

> So if you do not specify any rules, your code is not affected at all.
> If you specify rules, they get tested.

When the open PR https://github.com/ros-industrial/industrial_ci/pull/150 gets merged, then the above won't hold any more right?

>  As soon as projects start growing you have to take care of dependencies and package relations.

Well managing dependency doesn't require using `install` space does it?

If I sound as if I changed my poistion on this subject recently, I'm sorry. I might have done that a bit. I noticed adding instructions for `install` in CMakeLists.txt isn't that too difficult for me but I'm not sure that's the case for all ROS users.

ipa-mdl, 2017-03-26:
Perhaps the issue title is misleading.
We don't build directly to install space. We build to build/devel space and install the results to the install space (according to the rules).
The current behaviour is really odd:
1. build to build/devel space
2. run build test
3. clean
4.  build to build/devel space and install to install space
5. run rostest on install space

So in step 4 we are rebuilding everything from step 1, which doubles the time the build takes.

> Hm, I can't find where the install space is recommended

I don't use it locally. But as soon as others depend on my packages, I have make sure that the install space is set-up properly.
So I don't recommend to use (=source) your local install space, but to add install rules for other users.

On our robots all users chain to a common robot-specific install space that is chained to the global /opt/ros install space.

> Well managing dependency doesn't require using install space does it?

No, not directly. But the install rules make the build results explicit.
This way you can identify your build, exec and export dependencies more easily.

130s, 2017-04-07:
I posted my response https://github.com/ros-industrial/industrial_ci/pull/150#pullrequestreview-31649321, and that PR resolves this ticket I think.

Btw, now I'm convinced that recommending `install` space, even for novice users, is good, after [the discussing in buildsystem SIG](https://groups.google.com/forum/#!topic/ros-sig-buildsystem/OCtNc83s6GA).

closed 2017-04-07 by 130s