ros-industrial/industrial_ci #55: Opinion wanted for folding style

130s, 2016-05-27:
https://github.com/ros-industrial/industrial_ci/pull/48#issuecomment-219294989 thanks to @rhaschke:
> The output in the web interface is [cluttered](https://travis-ci.org/ros-planning/moveit_ros/builds/130311949) already in the overview (with `travis_time_start`, `set +x`, and `<<< >>>`. Hence, for MoveIt, I resumed to standard travis.yml files, improving them according to some inspiration gained from your script. I like the [result](https://travis-ci.org/ros-planning/moveit_ros/builds/130391657) much better - but that's my subjective view of course.

-  [This (say A](https://travis-ci.org/ros-planning/moveit_ros/builds/130311949) is what `industrial_ci` currently provides.
-  [B](https://travis-ci.org/ros-planning/moveit_ros/builds/130391657) is his version.

Which one do you like? For what reasons?

IMHO, indeed B looks fascinating, too. Removing non-essential lines (e.g. `set +x`) makes me feel good. On the other hand, I noticed that B visualizes more commands, while the A put together a chunk of commands into groups so that you can only expand the groups of your interest. I agree that which one is better is subjective.

gavanderhoorn, 2016-05-27:
As you say, highly subjective, but I think I like the conciseness of 'option B'. It shows at a glance which commands were used to setup the build env, which I think is nice.

Verboseness is good (as it allows you to see what really happened), and I feel that is important to keep around as a possibility. Being able to expand to drill down into details seems to allow for this, hence my preference for option B.

ipa-mdl, 2016-05-30:
I like the semantic  folding in 'A' , but dislike the hard-to-read `+++` prefixes.
So I would prefer the current folding with the verbose commands in 'B' style if unfolded.

Option 'B' might be easier to maintain..  and any travis user knows how to read it.

davetcoleman, 2016-07-07:
I like B much more - I can see the overarching commands without opening the folds

130s, 2018-07-11:
Can't point to specific PRs but this has been improved significantly. Closing.

closed 2018-07-11 by 130s