ros-industrial/industrial_ci #57: Customizable Docker script

130s, 2016-05-27:
I took one of the comments in https://github.com/ros-controls/ros_controllers/pull/224#issuecomment-222158639 as in that it'd be nicer if users can customize the script that runs `docker`.

VictorLamoine, 2017-01-03:
:+1:  For example if the build requires an external library (eg: `libcurl`) you can't use industrial_ci because there is no way to tell `industrial_ci` to install an APT package when it is running the docker ROS image.

Example package: https://github.com/InstitutMaupertuis/fanuc_post_processor

ipa-mdl, 2017-01-03:
>  there is no way to tell industrial_ci to install an APT package when it is running the docker ROS image.

You should use rosdep for this.

IMHO Docker customization should be used just for speedup only, like in https://github.com/ros-planning/moveit_ci
(see https://github.com/ros-industrial/industrial_ci/issues/75#issuecomment-267761781 as well)

VictorLamoine, 2017-01-03:
What if it is a source library? Then I think you are stuck.

For example: industrial_calibration requires ceres solver to be source compiled/installed, binary packages don't work.

Same applies if you need to install .deb files that you compiled to reduce build time, like in https://github.com/InstitutMaupertuis/fanuc_grinding, that' why I created https://github.com/InstitutMaupertuis/advanced_industrial_ci.

Running in containers is very cool but in that case we should be able to provide a script that runs within the container to manage our dependencies etc. (not 100% sure that is what the issue describes :smile:)

ipa-mdl, 2017-01-03:
> What if it is a source library? Then I think you are stuck.
> Same applies if you need to install .deb files that you compiled to reduce build time, 

BEFORE_SCRIPT should work in these cases.

130s, 2018-07-11:
Looks like questions are asked already. Overall I've been happy with how `industrial_ci` is flexible in this regard.

I'd also add specifically for adding upstream binary pkgs, `ADDITIONAL_DEBS` can be used ([doc](https://github.com/ros-industrial/industrial_ci/blob/53a1a5f232b2c6a4d6f92f8f790b779eab23a0c9/doc/index.rst#optional-environment-variables)).

closed 2018-07-11 by 130s