ros-industrial/industrial_ci #58: Run additional tests (non ROS) after build

gavanderhoorn, 2016-06-03:
I'm looking to include a `catkin_lint` step in the travis config at `ros-industrial/fanuc`.

Based on my understanding of the `industrial_ci` readme, I'm leaning towards including a custom `sh` script to run after the default `.ci_config/travis.sh`.

Is there something like an `after_script` (to complement `before_script`)? Or is my proposed approach ok? I'm assuming a non-zero exit status of any of the commands in my custom script will cause be build to fail?

shaun-edwards, 2016-06-03:
Should we include `catkin_lint`/other lint-ers in `industrial_ci`?

gavanderhoorn, 2016-06-03:
If configurable: maybe. I definitely don't think we want to do this by default, as some things linters report on could be considered style, which is rather subjective.

130s, 2016-06-03:
In terms of `catkin_lint`, +1 for having it as an option.

But I'm just not sure how we can achieve that; in my understanding it's a `CMakeLists.txt` macro that invokes catkin_lint or ros_lint and there's not really explicit shell-runnable commands. I'm hoping I'm missing something here.

gavanderhoorn, 2016-06-03:
[catkin_lint](https://github.com/fkie/catkin_lint) is a normal Python script.

shaun-edwards, 2016-06-03:
I had the same impression as @130s, but I guess I was wrong.  +1 to adding `catkin_lint` as an option.

ipa-mdl, 2016-06-04:
`ros_lint`  is CMake macro.. Every developer can add it to its packages, so we do not need special support for it in `industrial_ci`.

`catkin_lint` needs to be run externally, so we could add support for it:
* install it (python-catkin-lint)
* run it with configured warning level (before build or after build?)

gavanderhoorn, 2016-06-04:
I appreciate all the suggestions about `catkin_lint`, but my question was really broader than that. Do we use the `before_scripts` and friends, or is adding it to the `script` section also ok / acceptable.

130s, 2016-06-06:
> Is there something like an `after_script` (to complement `before_script`)? Or is my proposed approach ok? I'm assuming a non-zero exit status of any of the commands in my custom script will cause be build to fail?

With a risk of being pedantic, there's a section in [Travis CI's official doc](https://docs.travis-ci.com/user/customizing-the-build/#Breaking-the-Build) that says:

```
The `after_success`, `after_failure`, `after_script` and subsequent stages do not affect the the build result.
```

> Do we use the `before_scripts` and friends, or is adding it to the script section also ok / acceptable.

I regret to say that I personally haven't really taken much advantage of them, but I hope `industrial_ci` is made flexible enough so that you can utilize those as their intent.

And I guess for `catkin_lint` we got a "go" to integrate it as an option. Anyone interested in opening a PR for it might be assigned on this ticket?

130s, 2018-07-11:
- Couldn't spot where it was, but [AFTER_SCRIPT](https://github.com/ros-industrial/industrial_ci/blob/53a1a5f232b2c6a4d6f92f8f790b779eab23a0c9/doc/index.rst#optional-environment-variables) has been around for awhile in industrial_ci.
- Btw, `catkin_lint` option was added https://github.com/ros-industrial/industrial_ci/pull/219.

Closing but please feel free for more thoughts.

closed 2018-07-11 by 130s