ros-industrial/industrial_ci #63: Utilize ros.org as much as possible

130s, 2016-06-14:
Only recently I noticed that `ros.org` provides CI features for developers (*1). If there exists a group of people who knows everything about ROS development cycle, it's them -- OSRF. Why not delegate to what they provide as much as possible and let `industrial_ci` focus on what are not covered by them?

To do so we want to understand what stuff is available, and what to be delegated and what not. I started [a few question](http://answers.ros.org/question/236406/default-bool-value-for-test_commits-and-test_pull_requests-in-rosdistro-format/). Any input from anyone is appreciated as well.

*1 I thought I missed those when I initially started working on industrial_ci, but I thought I saw some comments at [June community mtg](http://rosindustrial.org/events/2016/6/14/june-2016-ros-i-community-meeting) (I couldn't keep the comment history before the meeting window shut down) that confirmed that the features from ros.org became available after industrial_ci kicked off.

davetcoleman, 2016-06-29:
Agreed. Does industrial_ci have a way to build ROS Kinetic on Ubuntu 16.04 or does it rely on Travis to support that? Travis said they won't support 16.04 for at least a year. The ROS buildfarm, however, uses Docker so can build Kinetic packages. I'm playing with the ROS buildfarm travis script [here](https://github.com/davetcoleman/rviz_visual_tools/pull/41)

gavanderhoorn, 2016-06-29:
Kinetic support was added in #52.

ipa-mdl, 2019-05-15:
@130s: Can we close this?

closed 2020-04-08 by ipa-mdl