ros-industrial/industrial_ci #64: ROS Prerelease Test option may not test what we want

130s, 2016-06-14:
Reported upstream https://github.com/ros-infrastructure/prerelease_website/issues/31
Can someone confirm this issue? If this holds then it's a shame that I didn't notice this...

130s, 2016-09-12:
[Confirmed in the upstream](https://github.com/ros-infrastructure/prerelease_website/issues/31):

> Currently, I believe it will always checkout code as defined in the `devel` section of the rosdistro distribution.yaml file.

This means the pull requests are not used in the prerelease test.

130s, 2016-09-12:
> This means the pull requests are not used in the prerelease test.

Add a temporary note https://github.com/ros-industrial/industrial_ci/pull/60/commits/a2a0e218416bcf58a7f414b9c3a594b3dde24204 in #60

ipa-mdl, 2016-12-15:
`generate_prerelease_script.py` has optional `--custom-branch` and `--custom-repo` parameters.

What do we want to test?
* for branch build: the branch
* for PR build: the remote branch?

ipa-mdl, 2016-12-15:
https://github.com/ipa-mdl/industrial_ci/commit/c875eacdc5ef05fdc6c4445c446065f939ae05d1 should do the trick, needs a source entry in ROS distribution.
Example: https://travis-ci.org/ipa-mdl/ros_canopen/builds/184373698

130s, 2016-12-19:
> `generate_prerelease_script.py` has optional --custom-branch and --custom-repo parameters.

Very good to know this! 

> https://github.com/ipa-mdl/industrial_ci/commit/c875eacdc5ef05fdc6c4445c446065f939ae05d1 should do the trick, needs a source entry in ROS distribution.

Would you open a PR?
Although I believe upstream https://github.com/ros-infrastructure/prerelease_website/issues/31 needs fixed, your workaround seems like a minimum hack against the existing upstream api that industrial_ci can take advantage from (and I think advantage exceeds disadvantage here).

> What do we want to test?
> 
> -    for branch build: the branch
> -    for PR build: the remote branch?

I've never thought of testing branches without PRs. Unless there's a need we can stick to PR only but I'm note opinionated here.

----

FYI just sharing a note of what I found about `--custom-repo`; it is used to specify [underlay](http://wiki.ros.org/catkin/Tutorials/workspace_overlaying#Chaining_catkin_workspaces) workspace. And its usage was a bit tricky so I opened https://github.com/ros-infrastructure/ros_buildfarm/pull/358 for improving help message.

130s, 2016-12-19:
We *could* implement a workaround using `--custom-repo` option for generate_prerelease_script.py, but because I'm afraid that's not what `underlay` intends, I prefer @ipa-mdl's https://github.com/ros-industrial/industrial_ci/issues/64#issuecomment-267469836.

ipa-mdl, 2016-12-19:
> ipa-mdl@c875eac should do the trick, needs a source entry in ROS distribution.
Would you open a PR?

It is based on #83, I will open a PR after it was merged :)

> Although I believe upstream ros-infrastructure/prerelease_website#31 needs fixed, your workaround seems like a minimum hack against the existing upstream api that industrial_ci can take advantage from (and I think advantage exceeds disadvantage here).

ros-infrastructure/prerelease_website#31 is not important for PR testing, since we provide the underlay code ourselves. I don't thinks it's a hack, `prerelease_clone_underlay.sh` already checks fo existing code.

> I've never thought of testing branches without PRs. Unless there's a need we can stick to PR only but I'm note opinionated here.

Testing the PR is the most intutive version, I wasn't sure if it is feasible, but it turned out be simple :)

PR #85 (2016-12-21): test prerelease from travis source
This patch runs prerelease tests on the job-bundled code (branch or merge) instead of the ROS distro devel branch.
For obvious reasons this does not work if another repository needs to be tested (i.e. `PRERELEASE_REPONAME` is set)
In this case the devel branch will be tested (as for the prerelease website).

Please note: The devel-job entry in ROS distro is still mandatory.

This might fix #64
closed 2016-12-21 by 130s