ros-industrial/industrial_ci #66: Reduce verbosity?

davetcoleman, 2016-06-21:
When using industrial_ci for MoveIt!'s many repos Travis is giving us the following warning:

> This log is too long to be displayed. Please reduce the verbosity of your build or download the raw log.

e.g. [here](https://travis-ci.org/davetcoleman/moveit/jobs/139272939). Any thoughts on how to optionally reduce output?

130s, 2016-06-21:
Yes this is a known issue.

There's a WIP https://github.com/ros-industrial/industrial_ci/pull/43 but as commented https://github.com/ros-industrial/industrial_ci/pull/43#issuecomment-223640058 the current approach isn't acceptable IMO.

I appreciate more ideas...

130s, 2017-02-14:
I guess we can close this after https://github.com/ros-industrial/industrial_ci/pull/94

closed 2017-02-14 by 130s