ros-industrial/industrial_ci #68: Use ROS official docker image

130s, 2016-06-29:
Suggested at https://github.com/ros-industrial/industrial_ci/issues#discussion_r68967262

I (or someone if interested) will be able to start working on this (before making any changes I want a refactoring PR #67 to be merged).

130s, 2016-06-29:
Suggested at https://github.com/ros-industrial/industrial_ci/pull/52#discussion_r68967262

I (or someone if interested) will be able to start working on this (before making any changes I want a refactoring PR #67 to be merged).

@davetcoleman @evenator

evenator, 2016-06-30:
I think the indigo-ros-core docker image is appropriate. I don't expect to have spare cycles for this in the immediate future, so it's up for grabs if anyone else wants to implement it.

Just swapping from the included dockerfile to the indigo-ros-core image is probably a minimal change. Even better would be to create an image specifically for Industrial CI on dockerhub.

Another option, perhaps the best, would be to figure out how to use the docker images that the build farm and prerelease jobs use. I personally find that the prerelease test is unbearably slow, but maybe someone could cut it down to something more manageable.

davetcoleman, 2016-07-05:
>  figure out how to use the docker images that the build farm and prerelease jobs use

I've started to switch [my Travis jobs](https://github.com/davetcoleman/rviz_visual_tools/blob/kinetic-devel/.travis.yml) to use the build farm CI, I think that's the best direction to go in actually.

asmodehn, 2017-08-03:
I also agree with this...

I recently switched to industrial_ci from my custom travis + shell script setup (just two files about 60 lines total, doing mostly the same thing), and overall I find industrial_ci more complicated than is required. 

This would be a good step to simplify the overall CI build.

ipa-mdl, 2017-08-03:
> and overall I find industrial_ci more complicated than is required.

It would be great if you could elaborate this (in a new issue).
This would help us to improve `industrial_ci`.

> This would be a good step to simplify the overall CI build.

Which step exactly?
The prerelease script can be executed with `PRERELEASE=true`.
However, currently we don't use the rosinstall file in this case.
In general the buildfarm script don't offer any customization options for the build process.

The ROS docker images can be used though ([example](https://github.com/ros-industrial/industrial_ci/blob/master/.travis.yml#L29)).

asmodehn, 2017-08-03:
I am currently running industrial_ci following the example linked form the README.
These imply a quite long `prepa_docker_image` step that is not useful in my case.
Me not knowing about DOCKER_IMAGE from the start might just be a missing example to follow using this feature.

On the other hand the docs are saying : 

> DOCKER_IMAGE (default: not set): Selects a Docker images different from default one. Please note, this disables the handling of ROS_REPOSITORY_PATH and ROS_DISTRO as ROS needs already to be installed in the image.

But I want to rely on shadow-fixed repo (which is not in the official ROS docker images). the reason is simple : I usually work on multiple package at once and I don't want to wait a distro sync to test new versions of two packages together.

I do that in my current .travis.yml like so  : 
```
before_install:
  # Getting docker ros image
  - docker pull ros:${ROS_DISTRO}-ros-core
  # Running as daemon
  - docker run --name ${CONTAINER_NAME} -d -t ros:${ROS_DISTRO}-ros-core /bin/bash | tee container.id
  # Switching to use ros shadow fixed packages
- docker exec -ti ${CONTAINER_NAME} /bin/bash -c "sed -i.bak -e s,packages.ros.org/ros/ubuntu,packages.ros.org/ros-shadow-fixed/ubuntu, /etc/apt/sources.list.d/ros-latest.list"
```

ipa-mdl, 2017-08-03:
You could use the ones from https://hub.docker.com/r/rosindustrial/ci/tags/ as well.
I plan to use them by default, but I have not implemented it yet.

asmodehn, 2017-08-04:
Oh I didn't know about these. Thanks !

asmodehn, 2017-08-05:
I am now using the ROS Industrial docker images. I encountered an unexpected issue : https://travis-ci.org/pyros-dev/pyros-utils/jobs/261293571

I am using the shadow fixed image, and the missing package is already in shadow fixed http://repositories.ros.org/status_page/ros_lunar_default.html?q=AlexV.
How come rosdep cannot get it ? Something wrong with that image ?

ipa-mdl, 2017-08-05:
> Something wrong with that image ?

Yes. It is not using shadow fixed, but the regular one. Will fix it.

ipa-mdl, 2017-08-06:
Fixed in https://github.com/ros-industrial/docker/pull/12

130s, 2018-07-11:
This was addressed in https://github.com/ros-industrial/industrial_ci/pull/52 and many subsequent PRs I believe.

closed 2018-07-11 by 130s