ros-industrial/industrial_ci #73: Should 'rosdep install' be run without '-r'?

gavanderhoorn, 2016-07-08:
I guess this is again a question of 'when to fail': `rosdep` is run with the `-r` argument by `rosdep-install.sh` at the moment. That makes it continue when it encounters a problem while resolving / installing dependencies, which seems like a good thing (sometimes).

In the case of `staubli_experimental` (example), this means that the build will fail at a (much) later point, as `staubli_experimental` depends on packages in `staubli`, but those haven't been released, and the `staubli` source repository is not included in the workspace setup for `industrial_ci`. `rosdep` cannot resolve the packages, but continues.

I don't know whether fail-early would be good here: how many times has `-r` saved a build that would've otherwise failed?

See [ros-industrial/staubli_experimental/builds/143354933](https://travis-ci.org/ros-industrial/staubli_experimental/builds/143354933) for a failed build.

130s, 2016-07-08:
Here are my subjective view for supporting fail-later (i.e. `rosdep` continues regardless resource-not-found error):
- IMHO, first 2 things we might want to check with CI among many others are build and test.
 - Wrong dependency definition doesn't always stop the system; for example, you can define infinitely many inexistent, unused run_depend in your package.xml. While `rosdep` fails to resolve those, build process won't be affected as long as necessary dependencies are resolved.
- Once resource issue is fixed then resource error won't be likely to happen (until new dependency definition is added).
- Missing resource is almost guranteed to be captured somewhere in the entire test process (rosdep, build, test).

Downside includes that jobs last longer, and there can be multiple errors so spotting them becomes harder (I guess this is the case with `staubli`?).

As a workaround(? not sure if this is correct wording here), how about this; 

- When there was resolution errors during `rosdep-install.sh`, the "fold" section turns to red, but the whole script goes on. This hopefully makes locating if error occurred during rosdep process a little easier.

Hm, not very good. Ideas welcomed.

ipa-mdl, 2017-01-12:
TL;DR: rosdep failures should stop the test immediately.

> you can define infinitely many inexistent, unused run_depend in your package.xml

This contradicts all ROS-I quality demands.
Most likely inexistent dependencies are the result of typos or outdated keys etc.

> Missing resource is almost guranteed to be captured somewhere in the entire test process 

The CI tests should better give a good indication which parts of the chain did fail.
If I made an error with the rosdep keys, I expect rosdep to fail. Not cmake and definitely not the compilation.

> Downside includes that jobs last longer, and there can be multiple errors so spotting them becomes harder

If rosdep will fail quite early in the process.
Contrary it might take even longer if we do not stop immediately, because 999 of 1000 will get installed before catkin will get a chance to complain.

130s, 2017-01-12:
+1 for fail-early. Over the time since my last post here I've become pickier about wrong dependency definition.

ipa-mdl, 2017-01-12:
this issue can be closed now :)

closed 2017-01-12 by ipa-mdl