ros-industrial/industrial_ci #76: Integration on GitLab CI

130s, 2016-07-19:
From @VictorLamoine at [ML](https://groups.google.com/d/msg/swri-ros-pkg-dev/xP8GJ0Dx_6M/co2pJMejBQAJ):

> Is there anything that would prevent me from adapting the existing scripts to GitLab CI (GitLab Runner)?
> I'm looking forward to use CI on private repositories hosted on GitLab and want to know if it is possible.

Running `industrial_ci` on non-Travis CI systems should be possible; main script is currently named as [travis.sh](https://github.com/ros-industrial/industrial_ci/blob/master/travis.sh), but should be designed to be agnostic from any particular CI system. However I haven't tested on other CIs so can't guarantee.

> Once (if) it works, how should we handle that, make a new branch in the GitHub industrial_ci repository?

In your CI config on Gitlab, you need to define the same environment variables and run `git clone`, as you do for Travis in your `.travis.yml`.

I'd be curious how things go with you, so feel free to ask further here if you face issues.

130s, 2016-08-22:
Closing for inactivity. Please feel free to reopen.

130s, 2016-09-06:
@VictorLamoine if you had a chance to integrate industrial_ci on GitLab, did it work?

VictorLamoine, 2016-09-06:
Not yet, but I will probably try next week.

I'm going to work in this public repository:
https://gitlab.com/VictorLamoine/gitlab-ci-ros-test

VictorLamoine, 2017-01-02:
I have made a working version of an `industrial_ci` equivalent for GitLab. It is much poorer in terms of functionalities.

https://gitlab.com/VictorLamoine/ros_gitlab_ci

Thanks to Docker this was pretty easy to do!

130s, 2017-01-03:
What I wanted to mean is that it would have been great if `industrial_ci` would be extended to accomodate new CI platform such as GitLab, rather than creating a whole new ci package, so that we can share the common codes and reduce maintenance effort that was/is made sporadically throughout the entire ROS community without particular purpose. And to emphasize that it's not limited to Travis CI,  I named the package to `ci` when I originally "forked" `jsk_travis` https://github.com/ros-industrial/ros_industrial_meetings/issues/1#issuecomment-157440992.

I understand that there are cases that creating new ci config makes sense for the needs (e.g. [moveit_ci](https://github.com/ros-planning/moveit_ci) was needed to handle their needs https://github.com/ros-industrial/industrial_ci/issues/75). I'm not experienced at all with it but are there specific requirements GitLab enforced you to create a new CI pkg @VictorLamoine ? I'm simply curious and if there's a chance of integrating the two I'll be happy to cooperate.

VictorLamoine, 2017-01-04:
I understand what you mean, it would be nice to be able to use the same CI repository with Travis, GitLab etc.

- First motivation was being able to work without bothering people managing this repository
- I wanted to learn how Docker works (never used it before)
- I first tried to run the original scripts but it was kind of a nightmare, the scripts here are very long and quite complex.
- The folding thing doesn't work as is in GitLab.
- I did not try installing ROS from apt like you do in your scripts (execpt from Kinetic)

I don't think there are any technical obstacles into integrating GitLab CI into these scripts.
Last but not least: I don't have much "free" time to work on these kind of things.

If someone has some time available to work on this I'll be happy to spend an hour explaining what I have done and how my repository works; but I can't integrate my current scripts inside this repository right now because I lack time.

130s, 2017-01-04:
Thanks for the explanation @VictorLamoine. Then I reopen this and leave it open to mark as a possible future-todo.

130s, 2017-03-26:
Closed via #98.

closed 2017-03-26 by 130s