ros-industrial/industrial_ci #87: How to deal with packages that /need/ to be installed (UPSTREAM_WORKSPACE)

gavanderhoorn, 2016-12-24:
Over at [ros-industrial/robotiq](https://github.com/ros-industrial/robotiq), we have a situation where a specific dependency (`soem`) hasn't been released into Jade, preventing us from running Travis builds for PRs against the `jade-devel` branch. No problem: use `UPSTREAM_WORKSPACE` and provide a `.rosinstall` file to install the dependency from source (ros-industrial/robotiq#88).

Unfortunately that doesn't work in this case, as the include directory layout of the installed package is rather different from the source space layout, leading to dependants running into `file not found` errors during compilation.

Short of getting `soem` released into Jade, what would be a way to deal with this using `industrial_ci`? Should we be looking into `before_script` and add an additional `.sh` script to setup a small workspace with just `soem` that we `catkin_make install` (or `catkin build` etc) and then overlay the main Travis workspace over the one providing `soem`?

130s, 2016-12-29:
Personally I don't have enough experience with the situation like this. If the case like this is common then we should add a functionality for that. If it's not then maybe I'm curious if the workaround you mentioned (using `before_script`) works.

ipa-mdl, 2016-12-29:
I would go for an external script with `before_script` or some travis entry  point like `install`.
What do you think of just installing the debian bundle (from indigo?) with `gdebi`?

> as the include directory layout of the installed package is rather different from the source space layout, leading to dependants running into file not found errors during compilation.

sounds like something is broken in the `soem` package, it should work for source/devel and for install space

ipa-mdl, 2019-05-15:
#361 implements a separate upstream workspace.
This issue seems to be resolved in the user repository, so I will close this one as well.

closed 2019-05-15 by ipa-mdl