ros-industrial/industrial_ci #89: C++11 is not enabled with Kinetic

VictorLamoine, 2016-12-30:
I'm migrating some packages from Indigo to Kinetic and Travis fails because of missing C++11.

Example pull request:
https://github.com/InstitutMaupertuis/simple_rviz_plugin/pull/3

Errors in the build:
https://travis-ci.org/InstitutMaupertuis/simple_rviz_plugin/jobs/187695811#L6273

The package compiles fine on my machine because I'm using gcc 6 which enables C++14 by default (https://gcc.gnu.org/gcc-6/changes.html).

The default gcc version on Travis is gcc 4.8.4 so the build fails. I think that `industrial_ci` should upgrade the compiler to gcc 6 whenever kinetic (or later versions of ROS) is used.

Another argument is that [REP 3](http://www.ros.org/reps/rep-0003.html) states that with Kinetic C+11 must be available.

ipa-mdl, 2016-12-30:
The tests for kinetic will not be run by travis directly, but in a Xenial-based Docker container..
According to your travis job gcc 5.4.0 is used.
```
[catkin_tools_prebuild:cmake] -- The C compiler identification is GNU 5.4.0
[catkin_tools_prebuild:cmake] -- The CXX compiler identification is GNU 5.4.0
```

ipa-mdl, 2016-12-30:
You are using moveit, so better have a look at https://github.com/ros-planning/moveit/issues/289

VictorLamoine, 2017-01-02:
I have installed gcc 6.2 before launching the build and it still fails;
https://github.com/VictorLamoine/simple_rviz_plugin/blob/8d6d8a900196128bc710bd30de8bb5c48cd0b1fe/.travis.yml

https://travis-ci.org/InstitutMaupertuis/simple_rviz_plugin/builds/188198224

I need `industrial_core` in my project so I clone it in my Travis  script but it does seem to be compiled with gcc 6.

Did anyone solve this issue? Help would be appreciated.

ipa-mdl, 2017-01-02:
> I have installed gcc 6.2 before launching the build and it still fails;

As pointed out above, the tests for kinetic are run in a separate Docker environment. You cannot install packages into it.
MoveIt switched to c++11 in kinetic, but does not export the required flags. ~~You have to set them in your package explicitly.~~

> I need industrial_core in my project so I clone it in my Travis script but it does seem to be compiled with gcc 6.

Ah, now I understand, what's wrong ^^
`industrial_trajectory_filters ` is not ready for kinetic (c++1 support..) yet.
If you do not need it, just remove the folder..

VictorLamoine, 2017-01-02:
Ah, looks like my brain didn't want to read that :smile:  I'll try to install gcc-6 within the docker or use another docker image if this is possible. I'll report my test results here.

Deleting the directory would be a temporary fix, this really isn't a "solution" !

ipa-mdl, 2017-01-02:
>  I'll try to install gcc-6 within the docker or use another docker image if this is possible. I'll report my test results here.

You would have to fork industrial_ci and have to add it to the Dockerfile.
IMHO this is the wrong approach, kinetic code needs to work with gcc 4.9 (http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021)
 
> Deleting the directory would be a temporary fix, this really isn't a "solution" !

The solution is to fix `industrial_core`.

VictorLamoine, 2017-01-02:
I fixed `industrial_ci` and cloned my fork instead of the official one, this is a clean/temporary solution until `industrial_core` is fixed, thanks for the help :+1: 

Closing as the issue is not related to `industrial_ci` but to `industrial_core`

closed 2017-01-02 by VictorLamoine