ros-industrial/industrial_ci #97: manifest.xml support needed?

ipa-mdl, 2017-01-12:
(from #95 https://github.com/ros-industrial/industrial_ci/pull/95#issuecomment-272205293)

`catkin` and `rosdep` prefer `manifest.xml` over `package.xml` in hybrid packages.
`rosdep` prefer  works at least for external dependencies. `catkin` falls back to `rosbuild`, which fails because it is not setup properly.

I guess the only reasonable fix is to delete all `manifest.xml` files and print a warning.
However, I don't even know if someone really keeps `manifest.xml` files around..

130s, 2018-07-11:
>  However, I don't even know if someone really keeps manifest.xml files around..

Same thoughts here, and removed `bug` label.

ipa-mdl, 2019-07-14:
looks like this is not needed.

closed 2019-07-14 by ipa-mdl