ros-industrial/industrial_ci #98: CI consolidation roadmap

ipa-mdl, 2017-01-14:
This issue proposes a roadmap for consolidating `industrial_ci` and [moveit_ci](https://github.com/ros-planning/moveit_ci) as discussed in #75.

Preparations in `industrial_ci`
- [ ] cleanup outdated code
- [ ] merge or dismiss all open PRs
- [ ] move all scripts except `travis.sh` into `src`of `industrial_ci` package (standard ROS layout)
- [ ] split scripts: ci_main.sh (entry point, interface checks, test branching), `tests/source_tests.sh`(default test), 
- [ ] factor out code into semantical functions in to-be-sourced scripts
- [ ] factor out `$BUILDER == catkin` branches into `builder/catkin.sh` and branch to it from `builder-sh`
- [ ] add support for externally provided tests
- [ ] replace `TRAVIS_*`variables by generic ones
- [ ] run all tests in docker
- [ ] add support for external docker images
- [ ] add more unit tests, remove/replace some existing tests for speed-up
- [ ] add `.travis.example.yml` for documentation purposes, `.travis.yml` is for testing the test system..

Until here is just refactoring. These changes do not affect behaviour

Pull in `moveit_ci`, keep history if possible
- [ ] create `fusion` branch
- [ ] migrate to `moveit_ci/utils.sh` as it is much cleaner ( includes travis-like folding!), use `ici_` prefix for functions
- [ ] add support for `ROS_REPO` and `TEST_BLACKLIST `
- [ ] integrate `check_clang_format.sh`, add option for clang version
- [ ] test with https://github.com/ros-planning/moveit

At this stage all tests should work *as-is* for `moveit_ci` and `industial_ci`, but with `move_it` folding
- [ ] decide new name and location if needed (TBD)
- [ ] make superseded repos use consolidated repo under the hood

New features for consolidated repo:
- [ ] introduce `scripts/run_test` with travis-like matrix lines for use without travis
- [ ] release as ROS package (distro-independent?)
- [ ] implement additional tests, e.g. `catkin_lint``
- [ ] add support for other CI services (#76)

@130s, @davetcoleman: what do you think?

130s, 2017-01-17:
Thanks for putting together!
Easier-to-use is something to think about too. https://github.com/ros-industrial/industrial_ci/issues/105

davetcoleman, 2017-02-15:
This looks good to me

130s, 2017-04-28:
ROS buildfarm testing is not "free" https://github.com/ros/rosdistro/pull/14820#issuecomment-298126877 and we should respect the effort of the maintainers there to minimize the server's cost, so making it easy to utilize online free service (e.g. Travis CI) is still very useful I think.

tylerjw, 2021-03-25:
>  test with https://github.com/ros-planning/moveit

you can probably check this one off... now that moveit is using industrial_ci (on the master branch for now, open pr's to old ros version branches).

closed 2021-07-21 by ipa-mdl