ros-industrial/industrial_ci #105: Add more sample config

130s, 2017-01-17:
Some complaints from active users, e.g. https://github.com/ros-industrial-consortium/bezier/issues/153

Indeed README gets necesarilly but surely complexed.

I'm also thinking of providing a template `.travis.yml` content so that users simply download in their `.travis.yml`.

ipa-mdl, 2017-01-17:
Pehaps we should split README.md into different chapters (-> doc directory).
And start with a `TL;DR` section with a minimal example.

VictorLamoine, 2017-01-17:
Complaint is a strong word. You have done some awesome work with this repository.
Please don't feel offended :hugs:

130s, 2017-01-18:
@VictorLamoine Thanks. I've seen "complaint" in OSS is often used as the same meaning as "feedback", and that was what I meant ;)

ipa-mdl, 2018-01-04:
A template might help preventing issues like #254

130s, 2018-02-05:
> -  provide a template .travis.yml content so that users simply download in their .travis.yml.

Template being added to https://github.com/ros-industrial/industrial_ci/pull/234

ipa-mdl, 2018-11-28:
The quickstart config is a little bit outdated and not really complete (#336).

130s, 2018-12-03:
> - [ ] [This link](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#optional-customize-catkin-config) is broken in `CATKIN_CONFIG`.

Can't find "optional-customize-catkin-config" section. And [customize-catkin-config](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#customize-catkin-config) section link is working. Checked.