ros-industrial/industrial_ci #106: Upstream/downstream logic design

ipa-mdl, 2017-01-17:
(branched off from #104)

The current logic needs to be improved, as it might not work as expected.

We have
* target repo with contained packages.
* a list of target to be tested (`TARGET_PKGS`)
* a rosinstall workspace (`UPSTREAM_WORKSPACE`)
* a list of downstream packages (`DOWNSTREAM_PKGS`)

We want to:
* install all dependencies that we do not have sources for (target or upstream or downstream)
* build all upstream dependencies (splitted from targets to identify problems with them)
* build all `TARGET_PKGS` (without dependencies) 
* run tests for `TARGET_PKGS`
* run installed tests for `DOWNSTREAM_PKGS` 
* build all `DOWNSTREAM_PKGS` (without dependencies)
* run tests for `DOWNSTREAM_PKGS`
* run all installed tests for `DOWNSTREAM_PKGS` 
* run installed tests for `DOWNSTREAM_PKGS` 

`DOWNSTREAM_PKGS` might be derived from `TARGET_PKGS`.
The list of upstream source depends can be easily derived from `TARGET_PKGS` and `DOWNSTREAM_PKGS`.

 We should discuss if we need to add a `DOWNSTREAM_WORKSPACE`.
I think this will include more problems than it solves as we have to deal with doubled packages.
Instead we might introduce a `ROSINSTALL_WORKSPACE` option as an alias for `UPSTREAM_WORKSPACE`.

In addition we have these issues:
- [ ] currently all dependencies get installed
- [ ] all packages from the rosinstall file get built per default
Both might take a lot of time.

ipa-mdl, 2017-02-20:
I have tried to summarize the types of packages:
[image]

ipa-mdl, 2017-03-07:
Actually, the different groups may overlap:
[image]

130s, 2017-07-05:
Sorry for taking forever to start looking into this. Just clarification for now,

> We want to:
> :
> - run installed tests for DOWNSTREAM_PKGS
> :
> - run tests for DOWNSTREAM_PKGS
> - run all installed tests for DOWNSTREAM_PKGS
> - run installed tests for DOWNSTREAM_PKGS

How is each item here different from others? Is this list sorted by a certain order?

>  The list of upstream source depends can be easily derived from TARGET_PKGS and DOWNSTREAM_PKGS.

Not making much sense to me.. Don't "upstream source depends" and `DOWNSTREAM_PKGS` contradict with each other do they?

ipa-mdl, 2017-07-06:
> How is each item here different from others? Is this list sorted by a certain order?

The list has the order of execution. The items separate the different stages and soruces to identify the source of failures.

> > The list of upstream source depends can be easily derived from TARGET_PKGS and DOWNSTREAM_PKGS.

> Not making much sense to me.. Don't "upstream source depends" and DOWNSTREAM_PKGS contradict with each other do they?

Here upstream means dependencies of target packages **and** downstream packages.
[upstream dependencies of target packages are as well upstream dependencies of downstream packages)
Upstream packages just get built, but not tested (for time saving, this could be an opt-out).
Downstream packages (=depend on one oft the target packgaes) gets built and tested.

Example:

target repositiory: https://github.com/ros-industrial/fanuc 
rosinstall:
 - https://github.com/ros-industrial/fanuc_experimental (downstream packages)
 - https://github.com/ros-planning/moveit (upstream packages)

This setup could be used to test;
 1. changes in `fanuc` do not break `fanuc_experimental`, especially launch file checks!
 2.  breaking changes in moveit

Right with the default config all of the packages get built.
And all depencies get installed..

With #137 only the required packages get resolved and built.

(I had brief look into the packages, the tests might fail because the `test_depends` as not properly configured)

gavanderhoorn, 2017-07-06:
@ipa-mdl wrote:
> (I had brief look into the packages, the tests might fail because the `test_depends` as not properly configured)

Is `the packages` referring to pkgs in `fanuc` and `fanuc_experimental`, or others?

ipa-mdl, 2017-07-06:
@gavanderhoorn:
> Is the packages referring to pkgs in fanuc and fanuc_experimental, or others?

yes, and most of the package format 2 packages ;)
The `_support` packages just contain `exec_depend`, but the roslaunch checks introduce `test_depend` as well.
In package format 1  `run_depend` meant `exec_depend` and `test_depend`.
This is the strict intepretation.
We could decide to treat `exec_depend` as `test_depend`  for the CI tests (perhaps an opt-in?).

gavanderhoorn, 2017-07-06:
Hm, interesting. That was not how I had interpreted the interplay between `test_depend` and `exec_depend`. I'd understood that the union of those two would basically be the pkgs that would be available during a test run, but `REP-140` does not say that explicitly.

If I understand you correctly we'd basically have to duplicate all the `exec_depend`s and add them as `test_depend`s? At least for the `_support` pkgs.

ipa-mdl, 2017-07-06:
As I said, this my rather strict interpretation, it might be too strict though.
But it does not hurt to add `test_depend` for all test dependencies.
And this differentation is could be helpful for speeding up the CI tests.
E.g. if a packages has `exec_depends` on `rviz`, but it does not need it for build and test then there is no reason to instal rviz (and X11).

In my implementation I include transitive `exec_depend`s as `test_depend`s.
This could be [adapted easily](https://github.com/ipa-mdl/industrial_ci/blob/enhancement/106/industrial_ci/src/deps.py#L71) to include all `exec_depend`s as  `test_depend`s.

That beeing said, I just came to the conclusion that the relaxed interpreation should be the default for practical reasons.
But I will keep the strict mode as an opt-in.

> If I understand you correctly we'd basically have to duplicate all the exec_depends and add them as test_depends? At least for the _support pkgs.

Yes, exactly.

gavanderhoorn, 2017-07-06:
I'm fine with the 'strict interpretation' actually. But I do agree that it's probably a good idea to have the relaxed version standard - as I can imagine a lot of devs assuming that that is actually the 'correct' way.

Unless of course your (and now mine) *strict interpretation* is actually the only correct one. But for that we'd need some confirmation from the OSRF / authors of the REP.

@dirk-thomas: could you comment on this?

dirk-thomas, 2017-07-06:
For the general approach of building one set of packages (since it contains changes) and afterwards building another set of packages on top of thos (to ensure that their tests continue to pass): this sounds a lot like "prerelease" jobs on the ROS buildfarm. My last years ROSCon presentation contain some detailed information about it. Please take a look a the diagram in the slides which describes the hierarchy pretty well: http://www.dirk-thomas.net/slides_roscon2016_ros-buildfarm/2016-10-09_roscon_ros-buildfarm.html#/7/2

Regarding the dependencies I will first comment on some of the statements made above:

> In package format 1 run_depend meant exec_depend and test_depend.

That is not entirely correct. With format 1 when building a package (with tests enabled) you need to only have the build dependencies installed. The test dependencies only list additional dependencies to run the tests. Therefore those dependencies are not needed to build a package. Also in format 1 the test dependencies only contain keys which are *not* already listed as build or run dependencies.

See the `test_depend` tag docs in REP 127 (http://www.ros.org/reps/rep-0127.html#test-depend-multiple).

The downside of this is that you have e.g. `gtest` listed as a build dependency even though you might not even want to configure any tests. That is the reason why in format 2 the semantic of the test dependencies was changed.

> We could decide to treat exec_depend as test_depend for the CI tests (perhaps an opt-in?).
> That was not how I had interpreted the interplay between test_depend and exec_depend. I'd understood that the union of those two would basically be the pkgs that would be available during a test run, but REP-140 does not say that explicitly.

Yes, that is how it is being handled on the ROS buildfarm. Before tests are being run the run as well as test dependencies are being installed.

> But it does not hurt to add test_depend for all test dependencies.

Yes, with format 2 there is no constraint (as it exists for format 1 (https://github.com/ros-infrastructure/catkin_pkg/blob/0ec1d3d30164aed656cbb4e823bbfffa836baae8/src/catkin_pkg/package.py#L523-L528)). But ... see next paragraph ...

> If I understand you correctly we'd basically have to duplicate all the exec_depends and add them as test_depends?
> And this differentiation could be helpful for speeding up the CI tests.
> E.g. if a package has exec_depends on rviz, but it does not need it for build and test then there is no reason to install rviz (and X11).

While this is certainly possible and I see why that would make sense you might want to consider how this is currently being used by the community. Since the ROS buildfarm uses the "relaxed" interpretation existing ROS packages will list their dependencies with that semantic and not duplicate their run dependencies as test dependencies if they are needed during testing. If you are going to apply the "strict" interpretation many existing packages will likely fail during testing since not all dependencies needed to run the tests are installed (since they are not listed as `test_depends`).

The "strict" interpretation could be considered in the future (for an upcoming format 3). I would like to mention though that the community has raised the concern in the past that they don't want to "list the same key multiple times". So I would expect some push back when this separation will be proposed since it will often require duplicating the run dependencies.

ipa-mdl, 2017-07-06:
Thanks @dirk-thomas!

So the relaxed definition is the way to go!
The strict definitions still makes sense for certain uses cases where the package maintainer / CI user takes care of the `test_depend`.

> > In package format 1 run_depend meant exec_depend and test_depend.

> That is not entirely correct. 

Yes, I was mistaken.
`run_depend` just resolves to `exec_depend` + `build_export_depend`.
I guss I was referring to the fact that "launch file check dependencies" had to be put in `run_depend` in package format 1.

dirk-thomas, 2017-07-06:
Since you had an extensive discussion about this please consider to contribute some clarifying paragraph to the REPs to avoid that kind of ambiguity in the future.

ipa-mdl, 2019-08-18:
#361 implements separate workspaces, which should be suffiicient.

closed 2019-08-18 by ipa-mdl