ros-industrial/industrial_ci #107: ROS-Install File Not Triggering

Jmeyer1292, 2017-01-17:
Greetings, I need help setting up industrial_ci for [Godel](https://github.com/ros-industrial-consortium/godel). Specifically, I'm trying to build Godel and its many source dependencies in my [own fork fist](https://github.com/Jmeyer1292/godel/tree/feature/add_ci).

I have enabled Travis and added a .travis.yml file to the following effect (taking mostly from moveit):
``` 
sudo: required 
dist: trusty 
language: generic 
compiler:
  - gcc
notifications:
  email:
    on_success: always
    on_failure: always
env:
  matrix:
    - ROS_DISTRO=kinetic  ROS_REPO=ros              UPSTREAM_WORKSPACE=godel.rosinstall
    - ROS_DISTRO=kinetic  ROS_REPO=ros-shadow-fixed UPSTREAM_WORKSPACE=godel.rosinstall
install:
  - git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
script: 
  - source .ci_config/travis.sh
#  - source ./travis.sh  # Enable this when you have a package-local script
```
[The build fails](https://travis-ci.org/Jmeyer1292/godel/jobs/192869969#L1892) indicating that I'm missing certain packages that should have been installed via the `godel.rosinstall` script mentioned above.

I know it's probably obvious, but what am I doing wrong here?

ipa-mdl, 2017-01-18:
Your syntax is not support yet (looks like `moveit_ci`, should be available with #98)

Please try:
```yaml
- ROS_DISTRO=kinetic  UPSTREAM_WORKSPACE=file ROSINSTALL_FILENAME=godel.rosinstall ROS_REPOSITORY_PATH=http://packages.ros.org/ros/ubuntu
- ROS_DISTRO=kinetic  UPSTREAM_WORKSPACE=file ROSINSTALL_FILENAME=godel.rosinstall ROS_REPOSITORY_PATH=http://packages.ros.org/ros-shadow-fixed/ubuntu
```
Or more condensed:
```yaml
env:
  global:
    - ROS_DISTRO=kinetic
    - UPSTREAM_WORKSPACE=file
    - ROSINSTALL_FILENAME=godel.rosinstall
  matrix:
    - ROS_REPOSITORY_PATH=http://packages.ros.org/ros/ubuntu
    - ROS_REPOSITORY_PATH=http://packages.ros.org/ros-shadow-fixed/ubuntu
```

Jmeyer1292, 2017-01-18:
@ipa-mdl Thank you! That seems to at least fix that issue. Now to try and fix all the other issues with my code.

I really appreciate it.

closed 2017-01-18 by Jmeyer1292