ros-industrial/industrial_ci #111: Everything was successful: your build exited with 1 - but Travis still says 'failed'

gavanderhoorn, 2017-01-20:
Afaict, builds for ros-industrial/abb#89 should be successful, and from the [log](https://s3.amazonaws.com/archive.travis-ci.org/jobs/193805112/log.txt) it would seem they are, but I'm seeing red crosses everywhere and Travis does not seem happy.

That makes me unhappy :cry: 

I'm not completely up-to-date, but I know you guys are working hard on improving `industrial_ci`. Could it be that one of the recently merged PRs inverted some logic somewhere?

gavanderhoorn, 2017-01-20:
Hm. Or is this:
```
[abb] Started testing...
find: `install/share/abb': No such file or directory
```
The cause of those failed builds?

Is this connected with #110?

ipa-mdl, 2017-01-20:
Should be resolved now, please restart your builds :)

ipa-mdl, 2017-01-20:
> Could it be that one of the recently merged PRs inverted some logic somewhere?

Nothing inverted here.
`industrial_ci` used to disable metapackages completely,
#108 removed this strange feature.
The install tests feature was not ready to handle metapackages, but this was not detected because of #101

ipa-mdl, 2017-01-20:
> Everything was successful: your build exited with 1

1 means failure in shell logic.
#100  and #103 improved the error propagation, it was broken for certain cases.

gavanderhoorn, 2017-01-20:
> 1 means failure in shell logic.

Yes, I know. Apparently the *Everything was successful* bit is not about the exit status of the build.

ipa-mdl, 2017-01-20:
So not everthing was successfull. Install tests did fail so it got marked red.
Apart from this bug in `industrial_ci`, builds will fail for more, previously undetected errors from now on.

>> 1 means failure in shell logic.

>Yes, I know

I have expected that :)

ipa-mdl, 2017-01-21:
Travis is happy again: https://travis-ci.org/ros-industrial/abb/builds/193867890

closed 2017-01-21 by ipa-mdl

gavanderhoorn, 2017-01-21:
Yep, thanks.