ros-industrial/industrial_ci #118: test_results block never reaches when NOT_TEST_INSTALL=false

130s, 2017-02-09:
During `catkin_run_tests` block, the [script terminates if tests return error](https://github.com/ros-industrial/industrial_ci/blob/c1594f46900f40a00697b5669acb5d062ffbdaf0/industrial_ci/src/tests/source_tests.sh#L137), so that `test_results` block, which comes later that runs `catkin_test_results --verbose` to show the error details, won't happen. Obviously this makes debugging hard.

Example project: [ar_track_alvar](https://github.com/sniekum/ar_track_alvar/blob/562fae16ec695a913d447e72951b6c65acd2b6b1/.travis.yml#L15), [result on Travis](https://travis-ci.org/sniekum/ar_track_alvar/jobs/199885756#L3336)

ipa-mdl, 2017-02-09:
I see two solutions:
1. append `|| true` to `catkin run_tests ..`
2. combine both lines: `catkin run_tests .. || catkin_test_results build`

130s, 2017-12-18:
This is becoming an issue for me now. 

@ipa-mdl thanks for rasing an attention in #250. Unless you've already started working on this, I'll try adding a patch very soon.

PR #251 (2017-12-18): print verbose catkin test results
fixes #118
closed 2017-12-18 by ipa-mdl