ros-industrial/industrial_ci #121: non-prerelease test fails with '/root/.ros/X/rosout.log': No such file or directoryError opening rosout log

130s, 2017-02-13:
Example https://github.com/sniekum/ar_track_alvar/pull/116 
Prerelease tests are passing.

130s, 2017-02-14:
Download file's storage location issue in https://github.com/sniekum/ar_track_alvar/pull/116 is resolved thanks to @ipa-mdl's help.

But it still doesn't make sense to me why the error message in the title `/root/.ros/X/rosout.log': No such file or directoryError opening rosout log` happened and why did it get resolved at the same time, because the solution in the same ticket was about where to download the resource needed for the tests, and that has nothing to do with the location of `rosout.log`'s location.

I won't, however, dig this deeper for now. Maybe those two locations are related somehow. We can reopen when this happens again.

130s, 2017-02-15:
Happening on another repo https://github.com/wg-perception/people/pull/49
An [example from Travis](https://travis-ci.org/wg-perception/people/jobs/202019286#L3209).

ipa-mdl, 2017-02-15:
I don't see a connection, at least not hint on read-only system problems..

130s, 2017-02-16:
Yeah seems like. 

I didn't mean to ask for the help @ipa-mdl -- I like to leave as much data as possible whenever I hit issues. But I always appreciate the company :)

closed 2017-02-16 by 130s

ipa-mdl, 2017-02-16:
It could have been a regression bug as well, so keep on pointing to problems :)

I am a little bit concerned that the log output is not as helpful as it should be..