ros-industrial/industrial_ci #122: Before Script: Read Only File System

Jmeyer1292, 2017-02-13:
I noticed today that a couple of build systems based on industrial_ci have started breaking where they previously worked.

See [this](https://travis-ci.org/ros-industrial/keyence_experimental/jobs/195682113) link for an example. I  clone, build, and install a git repo as part of my `before_script` but as of recently this step fails.

ipa-mdl, 2017-02-13:
Thanks for reporting!
I your case, #120 should solve the problem, but..

I would recommend to  switch to an absolut path for your clone/build  in `install-libsocket++.sh`
This way you don't interfere with the build in any way.
Only `$TARGET_REPO_PATH` is read-only.

That being said, I guess making `$TARGET_REPO_PATH` writable might be more user-friendly..
#121 is a result oft the read-only volume as well.

Jmeyer1292, 2017-02-14:
@ipa-mdl Can you elaborate on what absolute path is appropriate in this scenario?

ipa-mdl, 2017-02-14:
With #120 you script will be run with `/` as working directory (or `/root`, not 100% sure).

> Can you elaborate on what absolute path is appropriate in this scenario?

Anything except `$TARGET_REPO_PATH` and `$ICI_SRC_PATH`;)
Since all commands are run in docker (and as root :-/) you can use any path you like.
I would use `/tmp` or `/tmp/build_deps`.

In version 0.3.1 the before script was executed in `~/ros/ws_$TARGET_REPO_NAME/src`.

Which directory would you expect as a user?

Jmeyer1292, 2017-02-14:
So my initial logic was based off the travis CI script [examples](https://docs.travis-ci.com/user/installing-dependencies/). See bottom of the page.

In these, they just download a directory, unzip it, cd into it, and build right there.
I will change it to `/tmp` if I need to. I would prefer if I didn't have to modify anything. +1 to making that dir writable.

ipa-mdl, 2017-02-14:
> So my initial logic was based off the travis CI script examples

Hmm, they pull additional code right into the src repo folder..
It get's a little bit more complicated with running in docker..
The before script will be run in the default workdir of the  docker image as soon as #120 got merged
I wonder if we should set it to a more reasonable default than `/`.

> I would prefer if I didn't have to modify anything. 

So just wait for #120 (and/or #123) to be merged :D

Jmeyer1292, 2017-02-14:
Ok. I appreciate your support @ipa-mdl.

closed 2017-05-05 by Jmeyer1292