ros-industrial/industrial_ci #125: Use sh, not bash?

130s, 2017-02-14:
I open this as a placeholder for a discussion.

It'd be ideal for all scripts to accommodate more platforms to be run on. 
However, I don't know how critical using `sh` would be though. Even on Windows 10 `bash` runs on these days.

ipa-mdl, 2017-02-14:
Some features are supported by `sh`, especially the calculations and the `BASH_SOURCE` feature.
we should try to keep compatibility with bash and dash.
(I will add a shellcheck test after the refactoring)

ipa-mdl, 2019-05-15:
shellcheck compliance and test jobs gets added in #361 
We are using arrays a lot, which are not available in `sh`.

If needed, we could try to migrate the entry script to `sh`, but keep using bash inside Docker.

ipa-mdl, 2020-04-08:
At this point we cannot switch back to `sh`  anymore.

closed 2020-04-08 by ipa-mdl