ros-industrial/industrial_ci #126: cmake project support?

130s, 2017-02-14:
Just got a question if `industrial_ci` supports `cmake`, non catkin package (I assume something still related to ROS. E.g. [ROS QtCreator plugin](http://www.ros.org/news/2016/07/qt-creator-ide-plug-in-for-ros.html)) from @Levi-Armstrong.

I wouldn't say we would plan to support `cmake` project, at least for now, given industrial_ci is currently tied to `catkin_tools`. But I'll leave this open for discussion.

ipa-mdl, 2017-02-14:
If `catkin_tools` support plain cmake packages, `industrial_ci` supports it as well.

gavanderhoorn, 2017-02-14:
@ipa-mdl wrote:
> If `catkin_tools` support plain cmake packages [..]

according to the docs ([Workspace Mechanics - Source Packages and Dependencies](http://catkin-tools.readthedocs.io/en/latest/mechanics.html#source-packages-and-dependencies)) it does not.

ipa-mdl, 2017-02-14:
> according to the docs (Workspace Mechanics - Source Packages and Dependencies) it does not.

It should work for cmake project with proper package.xml.
The default source test does not work for for non-catkin packages.
What should get tested for bare cmake projects?

But it should be possible to choose or provide other tests (#98)

gavanderhoorn, 2017-02-14:
@ipa-mdl wrote:
> It should work for cmake projects with proper package.xml.

true, but I would not call that a 'plain cmake' package anymore.

`ament` changes this.

wxmerkt, 2019-04-21:
I have a plain-CMake project that is wrapped into ROS as an external library via a `package.xml` and `<buildtype>cmake</buildtype>`. Catkin builds it just fine and industrial-ci does so too, but it tries to test and of course can't find the output directories for `catkin_test_results ...` - hence it fails in PRERELEASE mode. Is there a workaround for this I am unaware of?

ipa-mdl, 2019-04-21:
>  hence it fails in PRERELEASE mode. Is there a workaround for this I am unaware of?

Prerelease just wraps https://github.com/ros-infrastructure/ros_buildfarm, not sure if it supports plain cmake packages

> can't find the output directories for catkin_test_results

This should not show an error.
catkin_test_results just crawls for xunit reports and dispays their errors found in there.
No xunit file, no error:
```
Summary: 0 tests, 0 errors, 0 failures, 0 skipped
```

wxmerkt, 2019-04-21:
I may have misinterpreted the travis result/error - the build I am referring to is [here](https://travis-ci.com/wxmerkt/apriltag/jobs/194231336)

ipa-mdl, 2019-04-21:
> I may have misinterpreted the travis result/error - the build I am referring to is here

No, you did not..
It looks like the the `ros_buildfarm` scripts passes a directory that is not exisiting.
`NOT_TEST_INSTALL` and `NOT_TEST_BUILD` are not used in prerelease test.
I have opened a new issue make this configurable (#362)

You could try the new version (#361), it uses the latest `ros_buildfarm` version from source,.
If the error persists, you shoud open an issue over there.

ipa-mdl, 2019-08-18:
`colcon` support plain `cmake` project, so it should work with the new `master` branch

closed 2019-08-18 by ipa-mdl