ros-industrial/industrial_ci #129: release management

ipa-mdl, 2017-02-17:
For now we have version tags for locking a certain version.
I would suggest to introduce a `stable` tag that gets updated once in a while with proper announcements.

This stable version should get release to ROS as well.

@130s, @evenator, @Jmeyer1292: Would this be an option for you to prevent problems like #122?

evenator, 2017-02-17:
:+1:  From me. (Although we might keep running master, because surprise features are fun.)

130s, 2017-02-17:
Would you elaborate the usecase from user's side? Trying to understand, I would just assume as follows: every once in a while when necessary,
- industrial_ci releases version `x.x.x-stable`. Or just `stable` without version name?
- users update their config (e.g. `.travis.yml`) to use x.x.x-stable.

I'm personally a fan of the current style, where users don't have to care about what's going on at industrial_ci at all once they introduced it to their repos (https://github.com/ros-industrial/industrial_ci/issues/122 is a rare exception). Since there are already 50+ user repos (generous guess I [presented](https://discourse.ros.org/t/ros-testing-continuous-integration-ci-and-deployment/1266)), we might want to keep the current cloning scheme. Maybe we want to:

- use `master` branch as stable.
- earlier adaptors can use an experimental branch.

So in the end, I'm probably catching up with the thought @ipa-mdl has in mind :)

ipa-mdl, 2017-02-17:
I just meant `stable` without version and it should be a tag and no branch (everything else is confusing).
I would keep master as "experimental" as it is tested with our unit tests only.
This should cover most use cases.

However, we cannot test all 50+ user repos for regression bugs (but extend our unit tests).
They are our public beta testers per default (my preference) :D
If we get no complaints for version `stable+1` for a certain time (say 1 month), we can update the tag after announcing it.

Whoever wants to opt-out can migrate to the `stable` tag.

I don't think that an `experimental` branch is necessary for now.

evenator, 2017-02-18:
I don't thinks tags should ever change. If it's going to move, even infrequently, it should be a branch.

ipa-mdl, 2017-02-18:
> I don't thinks tags should ever change.

After brief research: yeah, you're right. So `stable` branch it is :)

shaun-edwards, 2017-02-18:
+1 guys...and congrats on getting to a large enough number of users that this stuff now matters!  Great job!

closed 2021-07-21 by ipa-mdl