ros-industrial/industrial_ci #138: Introduce code analysis

shaun-edwards, 2017-03-07:
Does anybody have experience using [sonar qube](https://sonarqube.com/about).  It's a free code analysis service for open source projects.  It looks like it integrates nicely with [travis jobs](https://about.sonarqube.com/get-started/) as well.

130s, 2017-03-08:
I'm definitely interested in integrating something like sonar qube. This ticket can be labeled as enhancement. Related #98 #112

If anyone can start working please assign yourself or just claim here.

ipa-mdl, 2017-03-09:
According to the [travis docs](https://docs.travis-ci.com/user/sonarqube/) it is just a matter of setting up the authentication and simply run `sonar-scanner`. For this kind of use cases we just need to implement the test branching as pointed out in the roadmap.

It should work like specifying `TEST=sonarqube` which results in invoking the corresponding `sonarqube.sh` from the tests folder. We could even try to find a matching script in the target repository.

moriarty, 2017-12-15:
+1 for this.

I was introduced to Sonarqube when I joined a company over a year ago, and it's great.
I've brought it back to the university, where I was TA-ing a course, and had the students set up their own travis + sonar + github-sonar-plugin to automate as much of low level the review of their group projects... ( https://github.com/HBRS-MAAS/project )

It is easy to set up, if you're using python or java and a supported build system, either sonar cloud or the sonar docker image. I did have it running for PCL but it took a little more work for C++ you have to specify the include directories in the sonar.properties file.
https://github.com/SonarOpenCommunity/sonar-cxx/wiki/Code-checkers

That said, I haven't used it in combination with any ROS projects...
Which is why Google search brought me to this issue.

ipa-mdl, 2019-08-17:
support for `clang-tidy` will be added in #402