ros-industrial/industrial_ci #146: Using rosinstall, "Multiple packages found with the same name X"

130s, 2017-03-17:
E.g. [this result on Travis](https://travis-ci.org/ccny-ros-pkg/scan_tools/jobs/209873931#L232). Its [PR](https://github.com/ccny-ros-pkg/scan_tools/pull/52), and the [.rosinstall](https://github.com/ccny-ros-pkg/scan_tools/blob/20783d94fb9c2799f7edf7f702eef960c67ad3c9/.rosinstall) file, which does contain the repo where the PR is opened at.

```
ERROR: Rosdep experienced an error: Multiple packages found with the same name "laser_ortho_projector":
- src/ccny-ros-pkg/scan_tools/laser_ortho_projector
- src/scan_tools/laser_ortho_projector
Multiple packages found with the same name "laser_scan_matcher":
- src/ccny-ros-pkg/scan_tools/laser_scan_matcher
- src/scan_tools/laser_scan_matcher
Multiple packages found with the same name "laser_scan_sparsifier":
- src/ccny-ros-pkg/scan_tools/laser_scan_sparsifier
- src/scan_tools/laser_scan_sparsifier
Multiple packages found with the same name "laser_scan_splitter":
- src/ccny-ros-pkg/scan_tools/laser_scan_splitter
- src/scan_tools/laser_scan_splitter
Multiple packages found with the same name "ncd_parser":
- src/ccny-ros-pkg/scan_tools/ncd_parser
- src/scan_tools/ncd_parser
Multiple packages found with the same name "polar_scan_matcher":
- src/ccny-ros-pkg/scan_tools/polar_scan_matcher
- src/scan_tools/polar_scan_matcher
:
rosdep version: 0.11.5
```

ipa-mdl, 2017-03-17:
`local-name` must match the `$TARGET_REPO_NAME`  to get filtered out. This way it is independent from the fork.

130s, 2017-10-27:
See the discussion in https://github.com/ros-industrial/industrial_ci/pull/155
We as in industrial_ci team won't handle this issue as it's most likely? something that the users/developers need to take care of and CI can't help. Please add comment if there are still things that we can do.

closed 2017-10-27 by 130s