ros-industrial/industrial_ci #147: UPSTREAM_WORKSPACE syntax

ipa-mdl, 2017-03-18:
The current syntax of `UPSTREAM_WORKSPACE` along with `ROSINSTALL_FILENAME` is confusing.
I'd like to switch to an extended moveit_ci syntax:

UPSTREAM_WORKSPACE cases:
* `debian` (default): as is
* `file`: try
  * `ROSINSTALL_FILENAME `is set, try
     1.  `$ROSINSTALL_FILENAME.$ROS_DISTRO`
     1.  `$ROSINSTALL_FILENAME`
  *  otherwise try
     1. `.travis.rosinstall.$ROS_DISTRO`
     1. `.travis.rosinstall`
     1. `.ci.rosinstall.$ROS_DISTRO`
     1. `.ci.rosinstall`
     1. `.rosinstall.$ROS_DISTRO`
     1. `.rosinstall`
* `http*`: as is
*  otherwise, try
   1.  `$UPSTREAM_WORKSPACE.$ROS_DISTRO`
   2.  `$UPSTREAM_WORKSPACE.`
   3. fail

This effectively combines both variables.
In addition the new `file` paths can reuse existiing files and don't need a travis prefix

Open issues: 
* Do we want to support multiple workspaces?

ahoarau, 2017-03-22:
Multiple workspaces would be really great !

ipa-mdl, 2019-08-18:
A new systax was implemented in #361

closed 2019-08-18 by ipa-mdl