ros-industrial/industrial_ci #148: Deploy build results

ahoarau, 2017-03-20:
For deployment reasons, it would be great if we could send build results easily from the docker container up to the travis host, so we can take advantage of the ```deploy``` options : 

https://docs.travis-ci.com/user/deployment/releases/

is there a shared folder between the host and the container ?

ipa-mdl, 2017-03-20:
> is there a shared folder between the host and the container ?

No, not for the normal tests. The prerelease tests share a temporary folder with the host.

> For deployment reasons, it would be great if we could send build results easily from the docker container up to the travis host, so we can take advantage of the deploy options :

We could provide a `$DOCKER_RUN_OPTS` variable that you could use for setting up your volumes, e.g. `DOCKER_RUN_OPTS="-v $HOME/install:/root/catkin_ws/install"`.
Would that suit your needs?

ahoarau, 2017-03-20:
yes I think that would be perfect. But won't the dir will be deleted after ```source .ci_config/travis.sh```?

ipa-mdl, 2017-03-20:
> But won't the dir will be deleted after source .ci_config/travis.sh?

No, it should persist.

ipa-mdl, 2017-03-20:
please try #149

ahoarau, 2017-03-22:
Trying now, thanks !

ahoarau, 2017-03-22:
I'm getting this now : 
```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
[clean] Removing installspace: /root/catkin_ws/install
[clean] Failed to clean profile `default`
[clean] Note: Parts of the workspace have been cleaned which will necessitate re-configuring CMake on the next build.
```

ipa-mdl, 2017-03-22:
I guess the clean command tries to remove the install folder.
Please try https://github.com/ros-industrial/industrial_ci/pull/150 it does no need clean anymore.

ahoarau, 2017-03-22:
:+1: builds fine now

ahoarau, 2017-03-22:
It builds fine, but the result seems empty ! 

```
The command "source .ci_config/travis.sh" exited with 0.
 ls $HOME/install

The command "ls $HOME/install" exited with 0.
```

ipa-mdl, 2017-03-22:
#150 did not include #149 
I have rebased it, please try again.
My local test (with `run_ci`) worked :)

ahoarau, 2017-03-22:
working ! awesome thanks !

ipa-mdl, 2017-04-08:
I will close this one as #150 was merged.
Please do not forget to switch back to master :D

closed 2017-04-08 by ipa-mdl

ahoarau, 2017-04-08:
Excellent ! Thank you