ros-industrial/industrial_ci #163: Prerelease test on Xenial fails "/bin/sh: 1: locale-gen: not found"

130s, 2017-04-17:
For example in [ros-drivers/rgbd_launch/#38](https://github.com/ros-drivers/rgbd_launch/pull/38) (and [line on Travis CI](https://travis-ci.org/ros-drivers/rgbd_launch/jobs/222956588#L351)):

```
Step 4 : RUN locale-gen en_US.UTF-8
 ---> Running in 21573040740b
/bin/sh: 1: locale-gen: not found
Removing intermediate container 21573040740b
The command '/bin/sh -c locale-gen en_US.UTF-8' returned a non-zero code: 127
```

I almost believe this is related to https://github.com/ros-infrastructure/ros_buildfarm/issues/414, which is already closed. Not sure why this still occurs.

ipa-mdl, 2017-04-18:
> Not sure why this still occurs.

We are using the released version.
So we could switch to the the source version..
I'd like to refactor it anyway, because of #160, but not this week.

130s, 2017-04-24:
I see.
Looks like a fix is now available upstream (maybe https://github.com/ros/rosdistro/issues/14586#issuecomment-296302543). Now e.g. all prerelease test pass at [ros-drivers/openni_camera](https://travis-ci.org/ros-drivers/openni_camera/builds/225092148).

I close this but if there's a reason to keep it open please let me know.

closed 2017-04-24 by 130s