ros-industrial/industrial_ci #167: Docker: command not found

Jmeyer1292, 2017-05-05:
Greetings my friends in the industrial CI world. Once again I find myself with things that worked before that are now failing.

A number of packages, including `abb_experimental` and `keyence_experimental`, have started breaking with a command to the effect of:
```
/home/travis/build/ros-industrial/abb_experimental/.ci_config/industrial_ci/src/docker.sh: line 108: docker: command not found
```

See logs [here](https://travis-ci.org/ros-industrial/abb_experimental/jobs/228861494) and [here](https://travis-ci.org/ros-industrial/keyence_experimental/jobs/214380270).

At a quick gander, I see that the packages that are failing have `xenial` listed as their travis distribution, but that could be coincidence. Could you please shed some light on this?

130s, 2017-05-05:
So the quick fix I've opened to the mentioned repos seems to work.

Travis CI does not support Xenial as of today https://github.com/travis-ci/travis-ci/issues/5821#issuecomment-243617289.
- My guess is that Travis on those repo used to run on Ubuntu Precise, the default distro for Ubuntu, since `distro: xenial` was not resolved. 
- Travis CI [just updated their Precise image](https://blog.travis-ci.com/2017-05-04-precise-image-updates). Not clarified in the post though, I assume something changed regarding `Docker`.

130s, 2017-05-05:
Inspired by this issue, I opened #168 to update "documentation", which hopefully clarifies the current requirement of industrial_ci on Travis CI.

I'm closing this but please feel free to reopen if needed.
Thank you always for detailed reports and opinion @Jmeyer1292

closed 2017-05-05 by 130s

Jmeyer1292, 2017-05-05:
@130s I appreciate your work and your assistance a lot.

For something like this issue, where I had a rough idea of the issue, I'll still always try to make posts so that others who might have made the same mistake have some reference in the issue sections!

Let me know if I can help you out on developments going forward.

ipa-mdl, 2017-05-05:
"Fun" fact: Travis passes the distro without checks 
```
Build system information
Build language: generic
Build group: stable
Build dist: xenial
Build id: 228861489
Job id: 228861494
travis-build version: 66bd59b85
Build image provisioning date and time
Tue Apr 11 21:31:54 UTC 2017
Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 12.04.5 LTS
Release:	12.04
Codename:	precise
```
(https://github.com/travis-ci/travis-ci/issues/7296)