ros-industrial/industrial_ci #172: Add BUILD_PKGS_BLACKLIST?

gavanderhoorn, 2017-05-07:
This would be the opposite of `BUILD_PKGS_WHITELIST`.

Example use-case: `motoman_experimental` depends on `industrial_experimental`, for a single package only: `industrial_robot_pkg_gen`. As `industrial_experimental` has not been released, I'm using a `.travis.rosinstall` file. Everything works, but: `industrial_experimental` also contains `IRC_v2` packages, which run a lot of tests.

Testing is good, but `motoman_experimental` does not use anything from `IRCv2`, and I'd rather ignore the `IRCv2` packages completely in the Travis runs for `motoman_experimental`.

Instead of having to list each and every pkg I do want tested, it would be nice if we could invert the setting and use a blacklist.

ipa-mdl, 2017-05-07:
I have started to implemented this (other features) in https://github.com/ros-industrial/industrial_ci/pull/137

gavanderhoorn, 2017-05-07:
Ok, nice.

Would you want to keep this one open then, and close after implementation / merge of #137, or close right now as a sort-of duplicate?

ipa-mdl, 2017-05-07:
> Would you want to keep this one open 

Let's keep it open as a reminder

> and I'd rather ignore the IRCv2 packages completely in the Travis runs for motoman_experimental.

As a temporary fix you could create a `CATKIN_IGNORE` file in `BEFORE_SCRIPT` or run `catkin config ..` to blacklist it

gavanderhoorn, 2017-05-07:
Yeah, I tried to do that, but forgot that the `.rosinstall` file only gets processed by `travis.sh` itself. So my [line added before `travis.sh`](https://github.com/ros-industrial/motoman_experimental/blob/fa7237c2913f13b60697e53af2cacebd92ffc9f1/.travis.yml#L28) under `script` didn't work as I intended.

ipa-mdl, 2017-05-07:
> Does BEFORE_SCRIPT only work with actual scripts, or would a single script line (ie: touch ..) also work?

The string gets executed by `bash -c`, so `touch ...` works

130s, 2018-12-11:
>>    Does BEFORE_SCRIPT only work with actual scripts, or would a single script line (ie: touch ..) also work?
>
> The string gets executed by `bash -c`, so `touch ...` works

If I'm not mistaken, target repo is read-only so you can't `touch` into it (there fore something like https://github.com/ros-industrial/industrial_ci/pull/123 is useful)?

ipa-mdl, 2019-08-18:
`BUILD_PKGS_WHITELIST` was removed in #361 in favor of the new workspace management, which only supports "black-listing" by removing directories

closed 2019-08-18 by ipa-mdl