ros-industrial/industrial_ci #173: Enable Travis' cron job

130s, 2017-05-09:
Suggested by @ipa-mdl in the offline conversation.

+1 from me to enable it on this repo, and also recommend users (via mailinglist?) to enable it.

ipa-mdl, 2017-05-09:
Weekly or daily?

Daily would help to detect problems with our downstream dependencies earlier, but blocks build slots for the ´ros-indusstrial` organization, but I guess cron jobs will be exeucated on idle times.

ahoarau, 2017-05-09:
Going with nightly builds can't be bad for business 👍

ipa-mdl, 2017-05-09:
I have set-up a daily cron-job on master.
[image]

Let's try this for a couple of days :)

ipa-mdl, 2017-10-28:
cron jpbs are up and running.

closed 2017-10-28 by ipa-mdl