ros-industrial/industrial_ci #176: There's no way to NOT use install space

130s, 2017-05-15:
Opening as a separate ticket of what I poseted in https://github.com/ros-industrial/industrial_ci/pull/150#pullrequestreview-38160656.

Because `catkin config --install` is always run:
- there seems to be no way to NOT build into `install` space.
- If I'm right I want to add a feature to allow disabling `install` space.

Am I wrong?

This can cause an issue like https://github.com/ros-industrial-consortium/descartes/pull/198#issuecomment-301516448 where users want to test with the resources that are not deployed.
(For his case, I thought `NOT_TEST_INSTALL=true` could have worked but I was wrong; `NOT_TEST_INSTALL` just avoids running tests during the test steps for the `install` space.)

ipa-mdl, 2017-05-15:
> there seems to be no way to NOT build into install space.

For now `BEFORE_SCRIPT='catkin config -w $CATKIN_WORKSPACE --no-install'` is the way to go.

There are a couple of ways to make this more user-friendly:

1. remove `catkin config --install`, this would break the behaviour again, so :-1:
2. suppress `catkin config --install` with an opt-out flag like `USE_DEVEL_SPACE_ONLY`
3. `catkin config ${CATKIN_CONFIG:---install}`
4. `catkin config --install` followed by `if [ -n "$CATKIN_CONFIG" ]; then catkin config $CATKIN_CONFIG; fi`
5. suppressable `catkin config --install` followed by `if [ -n "$CATKIN_CONFIG" ]; then catkin config $CATKIN_CONFIG; fi` 

I'd prefer option 4, as a `CATKIN_CONFIG` could be beneficial for other use cases.
Option 5 is okay, but I'd like to avoid yet another specific flag.

130s, 2017-05-24:
Addressed in #177

closed 2017-05-24 by 130s