ros-industrial/industrial_ci #181: Cache compiled code with ccache

VictorLamoine, 2017-06-17:
The cache is saved/restored and allows faster build.
Very useful for big projects!

https://gitlab.com/VictorLamoine/ros_gitlab_ci/blob/master/ccache.bash
https://gitlab.com/VictorLamoine/ros_gitlab_ci/blob/master/.gitlab-ci.yml#L10-12

I'm pretty sure you can do the same with Travis CI.

ipa-mdl, 2017-06-18:
Thanks for pointing this out!

> I'm pretty sure you can do the same with Travis CI.

Wasn't too complicated: https://github.com/ipa-mdl/ros_canopen/blob/ccache/.travis.yml#L14
It saved 2 out of 6 minutes for `ros_canopen`.
After 2nd run: https://travis-ci.org/ipa-mdl/ros_canopen/builds/244099816

Right now I am not sure if we should just document this use case or even implement a `USE_CCACHE` flag.

ipa-mdl, 2017-06-18:
> Right now I am not sure if we should just document this use case or even implement a USE_CCACHE flag.

I have implemented a [feature](https://github.com/ipa-mdl/industrial_ci/tree/ccache) based on `CCACHE_DIR`, which integrates nicely with travis.
PR will follow as soon as travis backlosg is back to normal (4k jobs queued at the moment).

VictorLamoine, 2017-06-19:
Looks good :)

130s, 2017-06-29:
Addressed in #182

closed 2017-06-29 by 130s