ros-industrial/industrial_ci #183: Run pre/post scripts on docker-based industrial_ci container

130s, 2017-06-23:
Right now I think [pre/post-test scripts](https://github.com/ros-industrial/industrial_ci/blob/1306c2da7b4b4de657947d5d257004f2d1298c30/doc/index.rst#id33) run on the platform that is designated in the CI config (e.g. `.travis.yml`). This obviously means pre/post scripts can't run on the same platform where the tests run.

Is there a way with current setting to run pre/post scripts in the same docker container? Can't come up with a way.

Just as FYI...*An example usecase*: I have a package that bridges ROS and another framework where a server process runs. I want to install that framework (which requires some custom shell scripts so hard to get installed during `catkin` or any build) and get the server running in the pre script.

ipa-mdl, 2017-06-23:
> This obviously means pre/post scripts can't run on the same platform where the tests run.

The `BEFORE_SCRIPT` and `AFTER_SCRIPT` are run inside the container.

>  I want to install that framework (which requires some custom shell scripts

Just put the **invocation** in `BEFORE_SCRIPT`, e.g. `BEFORE_SCRIPT='./.ci_preprocess.sh'`.

130s, 2017-07-10:
> Just put the invocation in BEFORE_SCRIPT, e.g. BEFORE_SCRIPT='./.ci_preprocess.sh'.

Thanks, that works.

[This section](https://github.com/ros-industrial/industrial_ci/blob/1306c2da7b4b4de657947d5d257004f2d1298c30/doc/index.rst#run-pre-install-custom-commands) of current documentation talks about 2 ways, the one using `BEFORE_SCRIPT` variable and the one without and instead run your pre-test scripts before `.ci_config/travis.sh`. I assume this might have been written before the shift to Docker. Should we remove the latter?

ipa-mdl, 2017-07-11:
> Should we remove the latter?

Yes, otherwise it will confuse users.
We could even change the default example to `script: .ci_config/travis.sh`.

130s, 2017-07-11:
> Yes, otherwise it will confuse users.

That was me in https://github.com/ros-industrial/industrial_ci/issues/183#issue-238021537 :p 
I added a commit at https://github.com/ros-industrial/industrial_ci/pull/192/commits/581c3c3e3b3b743c21d969aeef68bebfd855bc89 in #192. Let us move the discussion there if any.

closed 2017-07-11 by 130s