ros-industrial/industrial_ci #186: Directory to clone industrial_ci into already exists in gitlab-ci

miguelprada, 2017-07-05:
I'm attempting to configure `industrial_ci` in a GitLab hosted project and I'm encountering some problems with the default `.gitlab-ci.yml` provided as an example.

I'm using the docker executor, and the problem I have is that if I clone `industrial_ci` as suggested in that config file, i.e. cloning into parent folder with

`git clone https://github.com/ros-industrial/industrial_ci ../.ci_config` 

then the build container complains that the directory already exists:

```
...
$ git clone https://github.com/ros-industrial/industrial_ci ../.ci_config
fatal: destination path '../.ci_config' already exists and is not an empty directory.
ERROR: Job failed: exit code 128
```

I guess the gitlab runner is mounting some volumes from the runner itself into the executor, and therefore the `../.ci_config` directory exists from previous jobs. In fact, by changing to some other path, the job succeeds on the first attempt but then fails in every subsequent one.

I can see that this path was introduced in #185 to fix the tests for `industrial_ci` itself, avoiding duplicate `industrial_ci` copies in the source directory, but I think that it is breaking the `for reguar users` use-case.

ipa-mdl, 2017-07-05:
Could you provide us (parts of) your config?
I can't imagine any reason why this folder should already exist.

ipa-mdl, 2017-07-05:
So it works for the shared runners at gitlab.com: https://gitlab.com/ipa-mdl/ci-example/pipelines
But I can confirm this problem with my local gitlab instance with a docker-based runner.

miguelprada, 2017-07-05:
This is my `.gitlab-ci.yaml`:

```yaml
image: docker:git # docker and git clients

services:
  - docker:dind # enable docker-in-docker

before_script:
  - apk add --update bash coreutils # install industrial_ci dependencies
  - git clone https://github.com/ros-industrial/industrial_ci ../.ci_config

indigo:
  script: ../.ci_config/gitlab.sh
  variables:
    ROS_DISTRO: "indigo"
    UPSTREAM_WORKSPACE: "file"
    ROSINSTALL_FILENAME: ".rosinstall"
    ROS_PARALLEL_JOBS: "-j1"
```

I'm running the gitlab runner as a docker image following [these instructions](https://docs.gitlab.com/runner/install/docker.html), and configured it as as project specific, and to use docker-based executors. The runner config is:

```
concurrent = 1
check_interval = 0

[[runners]]
  name = "roboticsrunner"
  url = <COMPANY_HOSTED_GITLAB_SERVER>
  token = <PROJECT_TOKEN>
  executor = "docker"
  [runners.docker]
    tls_verify = false
    image = "docker:git"
    privileged = true
    disable_cache = false
    volumes = ["/cache"]
    shm_size = 0
  [runners.cache]
```

miguelprada, 2017-07-05:
Also, by going over the logs in succesful runs, I can see that when I clone it at `.ci_config` instead of `../.ci_config` the folder persists as well, but is removed by the runner or the executor when fetching changes from the repo to be tested.

```
Running with gitlab-ci-multi-runner 9.3.0 (3df822b)
  on roboticsrunner (919d7a53)
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image docker:dind ID=sha256:5096e5a0cba00693905879b09e24a487dc244b56e8e15349fd5b71b432c6ec9f for docker service...
Waiting for services to be up and running...
Using docker image sha256:42d9b817e8b274dc8ab1c8d559a6339fd81084949287fbe2dc00574c6ba26732 for predefined container...
Pulling docker image docker:git ...
Using docker image docker:git ID=sha256:13455d9a5a879f7bfe41124a4c5eec4fdbef760d02e5600ba3f736a5c24e499d for build container...
Running on runner-919d7a53-project-19-concurrent-0 via 53e9f19ba5ec...
Fetching changes...
Removing .ci_config/
HEAD is now at 7a11b74 Set UPSTREAM_WORKSPACE to "file".
From <COMPANY_HOSTED_GITLAB_SERVER>/<GROUP>/<PROJECT>
   7a11b74..a83fc69  <BRANCH>  -> origin/<BRANCH>
Checking out a83fc69a as <BRANCH>...
...
```

ipa-mdl, 2017-07-05:
> Also, by going over the logs in succesful runs, I can see that when I clone it at .ci_config instead of ../.ci_config the folder persists as well, but is removed by the runner or the executor when fetching changes from the repo to be tested.

So it works with `.config`? Or only if the repository has changed?
I am thinking about adding `touch $DIR_THIS/CATKIN_IGNORE` to all CI script.

miguelprada, 2017-07-05:
It works when cloning into `.ci_config` irrespective of whether there have been changes to the repo or not (tested by manually retrying a finished job), since prior to fetching the directory is automatically removed. I guess the runner removes any untracked files before fetching.

ipa-mdl, 2017-07-05:
Actually, the problem with the duplicated packages gets fixed in https://github.com/ros-infrastructure/catkin_pkg/pull/183, no need to `touch CATKIN_IGNORE`.

gavanderhoorn, 2017-07-05:
That is some nice sleuthing @ipa-mdl 👍

PR #187 (2017-07-11): [gitlab] use .ci_config again
fixes #186 

@miguelprada: please review
closed 2017-07-11 by ipa-mdl