ros-industrial/industrial_ci #190: travis_wait support?

bmagyar, 2017-07-06:
One of our [new PR-s](https://github.com/ros-controls/ros_controllers/pull/286) over at `ros_control` is failing with [Travis giving up on it due to timeout](https://travis-ci.org/ros-controls/ros_controllers/jobs/250769760). On the PR you can see that the build is starting to fail once the new tests are added.

I tried the very naive approach of adding `travis_wait` to the source command but obviously it didn't do the trick. 

Have you experienced similar issues before?
https://travis-ci.org/ros-controls/ros_controllers/jobs/250811866

(The content of the PR is ok, it's a port from Indigo but I've also built & ran tests locally on Kinetic.)

ipa-mdl, 2017-07-06:
No yet..
`travis_wait 30 .ci_config/travis.sh`  should work.
(sourcing is neither required nor recommended anymore)

bmagyar, 2017-07-07:
Thanks, this did the trick!

closed 2017-07-07 by bmagyar