ros-industrial/industrial_ci #194: Build fails if no .rosinstall file is present

ipa-lth, 2017-07-13:
We might declare a rosinstall file to be necessary, but my personal opinion is, that if there is no rosinstall file, it should not install anything and just continue.
What is your opinion on this?

ipa-mdl, 2017-07-13:
> What is your opinion on this?

If a configuration is invalid, the build should fail,
e.g. to detect that someone forgot to commit the rosinstall file.

You could place an empty file in your repository.

ipa-lth, 2017-07-13:
I might missed it, but if you are consider a working repository (without the need for a rosinstall file) invalid, your README should be updated what valid means. In fact, I can just ask you, but someone else might cannot...

ipa-mdl, 2017-07-13:
> I might missed it, but if you are consider a working repository (without the need for a rosinstall file) invalid

If you don't need a rosinstall file, just keep the default `UPSTREAM_WORKSPACE=debian`.

https://github.com/ros-industrial/industrial_ci/issues/147 is related.

ipa-mdl, 2017-10-28:
@ipa-lth: Can we close this?

ipa-mdl, 2017-11-04:
It turned out that in an (outdated?) example `UPSTREAM_WORKSPACE=file` was included, which fails if no rosinstall file is present.
However, `UPSTREAM_WORKSPACE` can be omitted safely in this case.

closed 2017-11-04 by ipa-mdl