ros-industrial/industrial_ci #196: Release..

ipa-mdl, 2017-07-20:
I'd like to release `industrial_ci` as ROS package, this way it could be used locally.

Open issue: Do we need to release it for every distro or can we release it distro-independent? How?

@130s: what do you think?

130s, 2017-07-20:
+1

It was discussed long ago at https://github.com/ros-industrial/industrial_ci/issues/13. Back then we didn't have a way to run industrial_ci locally but [now we have it by calling `run_ci`](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#run-industrial-ci-on-local-host) (there seems to be [a user for it already](https://github.com/Nishida-Lab/pluginlib_arrayutil_client_ci/blob/89b934020931e0bbea6dde338d84ebb50190ef18/ci_local.sh#L2)). With that I suppose there can be usecases where having binary available of industrial_ci is convenient.

It'd also be a motivation for the maintainers (a.k.a me esp.) to tag a new release after changes are made where tagging a new version makes sense (I've been lazy in doing that).

> Open issue: Do we need to release it for every distro or can we release it distro-independent? How?

I think so.
And using the release feature on `ros.org`, I've never made a distro-independent release and I don't know if there's a way available to the end users like us. Worth asking on answers.

asmodehn, 2017-08-03:
My 2 cents : 
- ros package release is long and cumbersome, which can be justified for big and complex (C++) code, but not really for a set of simple scripts and tools.
- ros packaging means extra files, with different conventions that might not work with standard accepted by other communities (cf. python/catkin integration problems).

For a set of tools and scripts, I would just do a pip release (and maybe rewrite the code in python to be cross-platform - and avoid having to learn some Microsoft shell), actually similar to what others did for tools like bloom or catkin-tools. Much simpler, faster, cross-platform, and a better fit for tools overall.

ipa-mdl, 2017-08-03:
> ros package release is long and cumbersome

I cannot agree. It's quite easy in comparison to the steps that got automated under the hood.

> ros packaging means extra files, with different conventions that might not work with standard accepted by other communities

The package is already releasable, we just have to bloom it (which is easy).

> For a set of tools and scripts, I would just do a pip release (and maybe rewrite the code in python to be cross-platform - and avoid having to learn some Microsoft shell), actually similar to what [...]

I don't think that python scripts would be much simpler.
However, many tools we use are python-based with nice APIs to use, so it might be an option to port it.
But it would take a serious amount of time to do so and I don't that we would benefit from it at the moment.

asmodehn, 2017-08-04:
> > ros package release is long and cumbersome
> I cannot agree. It's quite easy in comparison to the steps that got automated under the hood.
My point exactly : for simple tools (not robot high performance critical software) we do not need all these steps. bloom-release take a fair amount of time, and a whole other repository.

But I do agree with the time required to port to python. I think the benefit will make it worth doing when non-linux platform will be targeted.

But then I am biased, I have been working for a while on https://github.com/asmodehn/pyros framework to do full dynamic ROS coding purely in python.

ipa-mdl, 2017-08-04:
The release should increase the accessibility.
So users can upgrade the package in their regular work flow.

> I think the benefit will make it worth doing when non-linux platform will be targeted.

We use docker and it supports only Linux natively.
Windows might be possible somehow, but it is a little bit pointless to target it (at the moment).

I have truly missed the flexibility of python for some features in `industrial_ci`, but I know that some tasks (e.g. job control) are easier in bash.

130s, 2018-07-11:
I'm also interested in porting to Python. Opened a ticket for that #301.

Back to the main point, is there anything holding off making a binary release?
IMO stuff like #279 #290 at least can be even more useful, as with the install binary there won't be a need for building and setting up the path to industrial_ci.

ipa-mdl, 2018-07-11:
> Back to the main point, is there anything holding off making a binary release?

Just the fact that we would have ro release the exact same code to different ROS distros,
As a python package we could distribute independently.

ipa-mdl, 2021-07-16:
I don't see this happening any time soon..

closed 2021-07-16 by ipa-mdl